UniversalRobots/Universal_Robots_ROS_Driver #298: Unable to find joint slide bars of rqt_joint_trajectory_controller

iakasoyr, 2020-12-07:
# Introduction to the issue
Dear members of this repo's communitiy,
I'm beginner of UR robots.
I downloaded and built the unversal_robot_ros_driver package.
I followed the setup process in Readme.md and confirmed ["Extract calibration information"](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#extract-calibration-information) normally .
However, when I tried "rosrun rqt_joint_trajectory_controller rqt_joint_trajectrory_controller", I couldn't find joint slide bars.
The terminal used for ur5e_bringup.launch did not print any error.
I probably misunderstand Remote Control mode on UR's teaching pendant and network settings.
So I would like to confirm correct meaning of the mode and settings.

# Versions
 - ROS Driver version: ROS Kinetic, ubuntu 16.04 LTS
 - Affected Robot Software Version(s): 5.4.2
 - Affected Robot Hardware Version(s): UR5e
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s):  external control-1.0.4.urcap

# Impact
I guess that my question and your answers will be useful for the readme documentation of setting and quick start.

# Issue details
I conformed that ping is correct response.
I finished setup and confirmed ["Extract calibration information"](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#extract-calibration-information) normally.
I couldn't find joint slide bars when I tried "rosrun rqt_joint_trajectory_controller rqt_joint_trajectrory_controller".
Also, the terminal for ur5e_bringup.launch does not show "Robot ready to receive control commands.".

## Use Case and Setup
 - ROS Driver version: ROS Kinetic, ubuntu 16.04 LTS
 - Affected Robot Software Version(s): 5.4.2
 - Affected Robot Hardware Version(s): UR5e
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s):  external control-1.0.4.urcap

Network setting:
ros PC's IP: 10.0.0.3
Robot's IP: 10.0.0.2
Host IP registerd in UR's teaching pendant: 10.0.0.3
Custom port in UR's teaching pendant: 50002 (default value)

Other setting:
Registered liscence of UR.
Turn on "Enable" of "Remote Control" in settings of UR's teaching pendant.

## Steps to Reproduce
(ROS PC) Installation of ROS Kinetic
(Robot) Preparation of real UR5e.
(Robot) Registration of the liscence.
(ROS PC) clone this repogitory.
(ROS PC) copy the external control file in "resources" directory to a usb stick.
(Robot) follow the [URCap installation](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/install_urcap_e_series.md) .
(ROS PC) Bulid this repogitory along with the [1st build method (not alternative)](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/install_urcap_e_series.md) 
(ROS PC) Extract calibration information.
(Robot) Tap a teaching pendant icon at the right top corner in the teaching pendant and change local control mode to remote control mode. 
(ROS PC)
```
roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=10.0.0.2 kinematics_config:=${HOME}/my_robot_calbration.yaml
```
-> could not confirm "Robot ready to receive control commands." on terminal.
(ROSPC)
```
rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller
```
-> could not see joint slide bars (same as [the image on previous issue](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/200) ).
"Controller manager ns" only showed "/controller_manager.
The Pulldown menu of "controller" did not have any candidate.

The terminal's output of "ur5e_bringup.launch" is following:
*Note* I did not real-time setup due to other projects.

NODES
  /
    controller_stopper (controller_stopper/node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ros_control_controller_spawner (controller_manager/spawner)
    ros_control_stopped_spawner (controller_manager/spawner)
    ur_hardware_interface (ur_robot_driver/ur_robot_driver_node)
  /ur_hardware_interface/
    ur_robot_state_helper (ur_robot_driver/robot_state_helper)

auto-starting new master
process[master]: started with pid [31957]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 34b8b5f0-388d-11eb-9751-54b2038bd999
process[rosout-1]: started with pid [31970]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [31987]
process[ur_hardware_interface-3]: started with pid [31988]
process[ros_control_controller_spawner-4]: started with pid [31993]
process[ros_control_stopped_spawner-5]: started with pid [32011]
process[controller_stopper-6]: started with pid [32023]
[ INFO] [1607346470.258530447]: Initializing dashboard client
process[ur_hardware_interface/ur_robot_state_helper-7]: started with pid [32041]
[ INFO] [1607346470.290116716]: Waiting for controller manager service to come up on /controller_manager/switch_controller
[ INFO] [1607346470.292902861]: waitForService: Service [/controller_manager/switch_controller] has not been advertised, waiting...
[ INFO] [1607346470.323478237]: Initializing urdriver
Warning: No realtime capabilities found. Consider using a realtime system for better performance
         at line 396 in /tmp/binarydeb/ros-kinetic-ur-client-library-0.1.1/include/ur_client_library/comm/pipeline.h
[INFO] [1607346470.521560]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1607346470.532430]: Controller Spawner: Waiting for service controller_manager/load_controller
Warning: No realtime capabilities found. Consider using a realtime system for better performance
         at line 396 in /tmp/binarydeb/ros-kinetic-ur-client-library-0.1.1/include/ur_client_library/comm/pipeline.h
Warning: No realtime capabilities found. Consider using a realtime system for better performance
         at line 396 in /tmp/binarydeb/ros-kinetic-ur-client-library-0.1.1/include/ur_client_library/comm/pipeline.h
[ INFO] [1607346472.918782393]: Loaded ur_robot_driver hardware_interface
[INFO] [1607346472.963045]: Controller Spawner: Waiting for service controller_manager/switch_controller
[ INFO] [1607346472.965392134]: waitForService: Service [/controller_manager/switch_controller] is now available.
[ INFO] [1607346472.965429670]: Service available.
[ INFO] [1607346472.965451792]: Waiting for controller list service to come up on /controller_manager/list_controllers
[ INFO] [1607346472.966416822]: Service available.
[ INFO] [1607346473.182912853]: Robot's safety mode is now NORMAL
[ INFO] [1607346473.183480294]: Robot mode is now POWER_OFF
[INFO] [1607346473.254867]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1607346473.256537]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1607346473.258271]: Loading controller: joint_state_controller
[INFO] [1607346473.265698]: Loading controller: scaled_pos_joint_traj_controller
[INFO] [1607346473.265980]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1607346473.268732]: Loading controller: pos_joint_traj_controller
[INFO] [1607346473.315782]: Loading controller: speed_scaling_state_controller
[INFO] [1607346473.353681]: Loading controller: joint_group_vel_controller
[INFO] [1607346473.357524]: Loading controller: force_torque_sensor_controller
[INFO] [1607346473.363738]: Controller Spawner: Loaded controllers: pos_joint_traj_controller, joint_group_vel_controller
[INFO] [1607346473.367641]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[INFO] [1607346473.369637]: Started controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller

## Expected Behavior
The joint slide bars are shown and move real UR5e by using the bars.

## Actual Behavior
The joint slide bars were not shown.

I will appreciate it if you could help me with the problem.
@fmauch @gavanderhoorn

chentongtong121, 2020-12-08:
hi   I have the same problem ， did you solve it already？

iakasoyr, 2020-12-08:
@chentongtong121 Unfortunately, No I didn't. If I will solve it, I will write the solution.

chentongtong121, 2020-12-08:
> @chentongtong121 Unfortunately, No I didn't. If I will solve it, I will write the solution.

my slide bars is ok now after closing the moveit

fmauch, 2020-12-09:
@chentongtong121 as you stated in the OP you didn't see the `Robot ready to receive control commands.` output in the terminal. Please see the [matching troubleshooting entry](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#i-started-everything-but-i-cannot-control-the-robot).

The README already contains the section

>Once the robot driver is started, load the previously generated program on the robot panel that will start the External Control program node and execute it. From that moment on the robot is fully functional. You can make use of the Pause function or even Stop (stop_button) the program. Simply press the Play button (arrow_forward) again and the ROS driver will reconnect.

chentongtong121, 2020-12-09:
> @chentongtong121 as you stated in the OP you didn't see the `Robot ready to receive control commands.` output in the terminal. Please see the [matching troubleshooting [...]

Thanks alot  I have solved it by playing External Control！！

iakasoyr, 2021-01-14:
> @chentongtong121 as you stated in the OP you didn't see the `Robot ready to receive control commands.` output in the terminal. Please see the [matching troubleshooting [...]

Thank you for your reply.
Your reply is true solution about this problem.

I misunderstood this sentence.
I tought that "the previously generated program" was already loaded when I finished the process indicated by the [Readme link](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#prepare-the-robot).
I should have pressed the Play button in the tablet of UR.

Anyway, thank you for your cooperation.
I closed the issue.

closed 2021-01-14 by iakasoyr

Haoran-Zhao, 2021-02-26:
> @chentongtong121 as you stated in the OP you didn't see the `Robot ready to receive control commands.` output in the terminal. Please see the [matching troubleshooting [...]

@fmauch I have setup the URcap and UR3 following the instruction. currently, I'm using Ubuntu 20.04 and noetic. The problem is after I execute the quick start command and press the play button of the external control program. I still didn't see "Robot ready to receive control commands. " What's the problem. Would you mind help me solve that? Thanks.

fmauch, 2021-03-01:
@Haoran-Zhao please do not reuse old issues for your own, new issue. Please open a new issue, so we can discuss there.

Haoran-Zhao, 2021-03-01:
@fmauch Thanks for your reply. I have create a new issue with the following link:
https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/326#issue-819133871