UniversalRobots/Universal_Robots_ROS_Driver #299: Could not get fresh data package from robot while  PROFINET is disabled

abdul-mannan-khan, 2020-12-09:
I have a another problem. My PROFINET is disabled still I am getting this error. 

Here is what I am doing:
```
$ roslaunch ur_calibration calibration_correction.launch \robot_ip:=192.168.0.100 \target_filename:="$(rospack find ur_calibration)/etc/ex-ur5_calibration.yaml"
$ roslaunch ur_robot_driver ur5_bringup.launch robot_ip:=192.168.0.100 [reverse_port:=REVERSE_PORT] \kinematics_config:=$(rospack find ur_calibration)/etc/ex-ur5_calibration.yaml
```

But after sometime, I get this error "Could not get fresh data package from robot". My PROFINET is disabled. Any suggestion please? Thank you for reading my query and your interest.

fmauch, 2020-12-09:
Could you please post the full log output of your second launch command?

Also: How is the (physical) connection between the robot and the ROS pc established? We recommend a direct cable connection.

gavanderhoorn, 2020-12-09:
Please also note the `\` in the commands shown. Not sure whether those are actual verbatim examples of the commands run, but those backslashes should not be there.

abdul-mannan-khan, 2020-12-14:
I was trying to connect UR5 through wifi. When I connected directly, the problem was solved. Thank you @fmauch. 

I have also removed `\` from my commands and the it was fine. Thank you @gavanderhoorn .

closed 2020-12-14 by abdul-mannan-khan