UniversalRobots/Universal_Robots_ROS_Driver #300: UR5e bringup.launch does not print Robot ready to receive control commands.

Hitdahit, 2020-12-17:
# Summary

Your issue may already be reported!
Please search on the [issue track](../) before creating one.

*Introduction to the issue*

# Versions
 - ROS Driver version: melodic, ubuntu 18.04
 - Affected Robot Software Version(s): 5.4.2
 - Affected Robot Hardware Version(s): ur5e
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s):external control-1.0.4.urcap

# Impact
I just followed this repo's README and how to setup e series(except real time settings. but i foloowed rest of instruction and there wasn't problem.). And I was able to launch bringup, but
it does not print "Robot ready to receive control commands".

# Issue details
When i launch bringup, It shows,
... logging to /home/mi2rl/.ros/log/67f7d0b8-3a86-11eb-9178-60eb69e0f981/roslaunch-mi2rl-labtop-2177.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/mi2rl/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://172.24.180.83:44919/

SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 500
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /hardware_control_loop/loop_hz: 500
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 500
 * /joint_state_controller/type: joint_state_contr...
 * /pos_joint_traj_controller/action_monitor_rate: 20
 * /pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/goal_time: 0.6
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_joint_traj_controller/state_publish_rate: 500
 * /pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /pos_joint_traj_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /robot_status_controller/handle_name: industrial_robot_...
 * /robot_status_controller/publish_rate: 10
 * /robot_status_controller/type: industrial_robot_...
 * /rosdistro: melodic
 * /rosversion: 1.14.9
 * /scaled_pos_joint_traj_controller/action_monitor_rate: 20
 * /scaled_pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
[... 199 lines cut ...]
         at line 396 in /tmp/binarydeb/ros-melodic-ur-client-library-0.1.1/include/ur_client_library/comm/pipeline.h
[ INFO] [1608167949.108079819]: Loaded ur_robot_driver hardware_interface
[ INFO] [1608167949.161849902]: waitForService: Service [/controller_manager/switch_controller] is now available.
[ INFO] [1608167949.161945627]: Service available.
[ INFO] [1608167949.162028678]: Waiting for controller list service to come up on /controller_manager/list_controllers
[ INFO] [1608167949.163641497]: Service available.
[INFO] [1608167949.280391]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1608167949.286796]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1608167949.287546]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1608167949.290931]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1608167949.291844]: Loading controller: joint_state_controller
[INFO] [1608167949.294797]: Loading controller: pos_joint_traj_controller
[INFO] [1608167949.302497]: Loading controller: scaled_pos_joint_traj_controller
[INFO] [1608167949.340827]: Loading controller: joint_group_vel_controller
[ INFO] [1608167949.357742206]: Robot's safety mode is now NORMAL
[ INFO] [1608167949.359345693]: Robot mode is now RUNNING
[INFO] [1608167949.394628]: Loading controller: speed_scaling_state_controller
[INFO] [1608167949.404646]: Controller Spawner: Loaded controllers: pos_joint_traj_controller, joint_group_vel_controller
[INFO] [1608167949.410370]: Loading controller: force_torque_sensor_controller
[INFO] [1608167949.418495]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[INFO] [1608167949.424523]: Started controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
and when i execute external control in teach pendant, 
[ INFO] [1608167967.986724346]: Robot requested program
[ INFO] [1608167967.986966329]: Sent program to robot
just this two logs showup and then there are no progress. I want what is problem in this situation.

fmauch, 2020-12-31:
In the future, please use [code blocks](https://docs.github.com/en/free-pro-team@latest/github/writing-on-github/creating-and-highlighting-code-blocks) to show console output. That makes it a lot easier to read.

Regarding your issue, I'd have to have a look at the generated code. You can extract it as described [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/74#issuecomment-571949829)

Hitdahit, 2021-01-05:
thank you for answering. here is your requested files
[filesur5e.zip](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/5768146/filesur5e.zip)

fmauch, 2021-01-05:
At a first glance, everything looks alright. Is the IP address `172.24.180.83` the same as you entered inside the URCap Installation configuration dialog?

Basically, what is missing (from my current understanding) is that the driver receives a keepalive signal (in [this line](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/resources/ros_control.urscript#L106). This could result from an atypical network setup such as with multiple virtual machines. Basically, the IP address passed to the script code for the reverse socket is deducted from the `RTDEClient`'s IP address [here](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/src/ur/ur_driver.cpp#L87). We had a discussion about this in #268. If you use a different IP address to communicate from the robot to your ROS machine, this could be the same issue. 

If this is not the case, you'd have to check your port settings on your ROS machine and make sure any firewall is disabled / configured to allow connections to this port.

Hitdahit, 2021-01-06:
Thank you so much.
I checked my laptop's firewall and it was activated.
So i simply deactivated laptop and after few more troubleshooting, I was able to move UR.

Thanks.

closed 2021-01-06 by Hitdahit