UniversalRobots/Universal_Robots_ROS_Driver #301: confusion about IP address for control

abdul-mannan-khan, 2020-12-20:
First of all, thank you all for all of your efforts in repository. I highly appreciate it. 

I am facing many problems related to this section. 

Question 1: My question is, should I use 192.168.56.101 or 192.168.0.100? If we have to use 192.168.56.101, why are we configuring IP address 192.168.56.1 in URCap (as described [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/install_urcap_cb3.md)) and why are using 192.168.56.101 and not 192.168.56.1? Please, read details below so that my problem could be cleared. I appreciate for your precious time. Thank you. Here is detail. 

I am confused about IP address as described [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#setting-up-a-ur-robot-for-ur_robot_driver) that I need to run the following to get configuration file

` $ roslaunch ur_calibration calibration_correction.launch robot_ip:=<robot_ip> target_filename:="${HOME}/my_robot_calibration.yaml"`

UR5 Network IP in Touch Panel is set to be 192.168.0.100. I replace <robot_ip> with that IP and it works. Next, I installed URCap as described [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/install_urcap_cb3.md) where I set the IP address: 192.168.56.1. However, as described [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#setting-up-a-ur-robot-for-ur_robot_driver) in **Quick start** 

> Once the driver is built and the externalcontrol URCap is installed on the robot, you are good to go ahead starting the driver. (Note: We do recommend, though, to extract your robot's calibration first.)
> 
> To actually start the robot driver use one of the existing launch files
> 
> $ roslaunch ur_robot_driver <robot_type>_bringup.launch robot_ip:=192.168.56.101

If I run
`$ roslaunch ur_calibration calibration_correction.launch robot_ip:=192.168.56.101 target_filename:="${HOME}/my_robot_calibration.yaml"`

I get the following warnings 
```
[WARN] [1607997530.263981]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[WARN] [1607997530.270256]: Controller Spawner couldn't find the expected controller_manager ROS interface.
```
However, I do not get any warnings if I run 
`$ roslaunch ur_calibration calibration_correction.launch robot_ip:=192.168.0.100 target_filename:="${HOME}/my_robot_calibration.yaml"`. 

Also my actual robot configuration and configuration in RViz is same.  if I run after this command, these commands after the one mentioned just before:
```
$ roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch limited:=true
$ roslaunch ur5_moveit_config moveit_rviz.launch config:=true
```
However, with this command, 
`$ roslaunch ur_calibration calibration_correction.launch robot_ip:=192.168.56.101 target_filename:="${HOME}/my_robot_calibration.yaml"`
my robot in RViz just lies the ground. 

**My question is should I use 192.168.56.101 or 192.168.0.100? If we have to use 192.168.56.101, why are we configuring IP address 192.168.56.1 in URCap (as described [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/install_urcap_cb3.md)) and why are using 192.168.56.101 and not 192.168.56.1?** Thank you for reading my question. 

Question 2: I believe that my second problem is due to the first question. I receive the error message as described in your blog [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#when-starting-the-program-on-the-tp-i-get-an-error-the-connection-to-the-remote-pc-could-not-be-established). In particular, I receive the error message on my Touch Panel as 

> The connection to the remote PC at 192.168.56.1:50002 could not be established. Reason: No route to host. 

I believe, if you could answer my Question 1, I would be able to solve this problem. I have tried all IPs. 192.168.56.1, 192.168.0.100, 192.168.56.101.  

Many thanks for your response.

abdul-mannan-khan, 2020-12-29:
OK. There are two many blogs and minor details. I have eventually figured it out. For everybody's help, I have made a video and uploaded. [https://www.youtube.com/watch?v=BS6pFmr7_lA&feature=youtu.be&ab_channel=Dr.Abdul-MannanKhan]

Feel free to comment for the improvement. Thanks. Here is the link.

closed 2020-12-29 by abdul-mannan-khan

gavanderhoorn, 2020-12-29:
Would be greatly appreciated if you could summarise what you figured out in a comment here.

That would save everyone from having to watch a video.