UniversalRobots/Universal_Robots_ROS_Driver #302: Incorrect joint states(Zero) publised intermittently

gokul-gokz, 2020-12-22:
# Summary

*Introduction to the issue*

# Versions
 - ROS Driver version: Ros Kinetic
 - Affected Robot Software Version(s): 3.10.0.0.76181
 - Affected Robot Hardware Version(s):

 - Robot Serial Number: 2014350137
 - UR+ product(s) installed:
 - URCaps Software version(s):

# Impact
Severe safety issue. Incorrect joint values result in unexpected behavior of the robot.

# Issue details
The joint states of the robot published by the joint state publisher are incorrect. The joint state publisher intermittently publishes zero joint state even-though the robot is not at that position. This results in unexpected behavior while executing a motion plan such as failure to execute the motion plan & sudden jerks while using velocity based control. 

Things tried so far:
1. Inorder to verify whether the issue is not caused by the robot, I checked the joint state values given by UR script commands for a fixed time interval without moving the robot. There were no zero joint values published. All the values were correct throughout the time period.

2. I also closed all nodes and ran only the robot driver and moveit. But still the issue exists.  

## Use Case and Setup
The setup is simple, just an UR5 robot with a robotiq end effector.

## Project status at point of discovered
I discovered this issue during normal use.

## Steps to Reproduce
1.  Run the universal robot ros driver.
2.  Run the moveit demo.launch 

## Expected Behavior
1. The joint state values published on the /joint_states topic should be exact as per the actual values. 

## Actual Behavior
1. The joint state publisher publishes zero joint state intermittently. I have attached a video of the RVIZ visualization.
https://user-images.githubusercontent.com/10492179/102939264-d242a980-4462-11eb-949b-b451271b6df1.mp4

## Workaround Suggestion
*If a workaround has been found, you are welcome to share.*

gavanderhoorn, 2020-12-23:
I'm not @fmauch, but I guess a `.bag` recorded while everything is running and the output of `rostopic list` and `rostopic info /joint_states` after you've started things *and* once after you've observed an "all-zeros `JointState` message would be more informative than a video.

you may also want to include whatever `rqt_graph` shows you.

gokul-gokz, 2020-12-23:
I was able to figure out the problem. 
I was re using the demo.launch by setting the fake execution args false. But i didn't notice there was a fake joint state publisher present in the demo.launch. I removed it and it's publishing properly. Thanks for pointing it out @gavanderhoorn .

closed 2020-12-23 by gokul-gokz