UniversalRobots/Universal_Robots_ROS_Driver #303: unused args [kinematics_config]

hustfzt2, 2020-12-24:
when I run: roslaunch ur_robot_driver ur5e_bringup.launch, the following issues are:
unused args [kinematics_config] for include of [/home/fzt/catkin_ws/src/universal_robot/ur_e_description/launch/ur5e_upload.launch]
The traceback for the exception was written to the log file.

gavanderhoorn, 2020-12-24:
Did you follow the readme, especially the [Building](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/tree/783b1d1f5774220058d0a2aa3dd93185641e8c7e#building) section?

It's important to `clone` the correct repositories.

fmauch, 2021-05-31:
Closing as a proper solution was proposed. Feel free to reopen or comment if the problem still persists.

closed 2021-05-31 by fmauch