UniversalRobots/Universal_Robots_ROS_Driver #304: cannot accept the action goal, the controller is not running.

hustfzt2, 2020-12-31:
when I run: roslaunch ur_robot_driver ur5e_bringup.launch,it works well,
however, when i run test file with the code as: 

#!/usr/bin/env python
from __future__ import print_function
import time
import roslib; roslib.load_manifest('ur_driver')
import rospy
import actionlib
from control_msgs.msg import *
from trajectory_msgs.msg import *

JOINT_NAMES = ['shoulder_pan_joint', 'shoulder_lift_joint', 'elbow_joint',
               'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']
Q1 = [2.2,0,-1.57,0,0,0]
Q2 = [1.5,0,-1.57,0,0,0]
Q3 = [1.5,-0.2,-1.57,0,0,0]

client = None 
def move_repeated():
    g = FollowJointTrajectoryGoal()
    g.trajectory = JointTrajectory()
    g.trajectory.joint_names = JOINT_NAMES
    
    d = 2.0
    g.trajectory.points = []
    for i in range(10):
        g.trajectory.points.append(
            JointTrajectoryPoint(positions=Q1, velocities=[0]*6, time_from_start=rospy.Duration(d)))
        d += 1
        g.trajectory.points.append(
            JointTrajectoryPoint(positions=Q2, velocities=[0]*6, time_from_start=rospy.Duration(d)))
        d += 1
        g.trajectory.points.append(
            JointTrajectoryPoint(positions=Q3, velocities=[0]*6, time_from_start=rospy.Duration(d)))
        d += 2
    client.send_goal(g)
    try:
        client.wait_for_result()
    except KeyboardInterrupt:
        client.cancel_goal()
        raise
def main():
    global client
    try:
        rospy.init_node("test_move", anonymous=True, disable_signals=True)
        client = actionlib.SimpleActionClient('/pos_joint_traj_controller/follow_joint_trajectory', FollowJointTrajectoryAction)
        print("Waiting for server...")
        client.wait_for_server()
        print("Connected to server")
        #move1()
        move_repeated()
        #move_disordered()
        #move_interrupt()
    except KeyboardInterrupt:
        rospy.signal_shutdown("KeyboardInterrupt")
        raise

if __name__ == '__main__': main()

the robot cannot move, and it occurs the errors in the terminator, "ur5e_bringup.launch", as:
cannot accept the action goal, the controller is not running.

fmauch, 2020-12-31:
The problematic line in your test code is

```
client = actionlib.SimpleActionClient('/pos_joint_traj_controller/follow_joint_trajectory', FollowJointTrajectoryAction)
```

The action server by default is on `/scaled_pos_joint_traj_controller/follow_joint_trajectory` as explained in the [Quick Start section](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#quick-start)

hustfzt2, 2021-01-04:
Thanks for your reply, when I tested the action server, it occurs the same errors, could you please share one test file, many thanks.

gavanderhoorn, 2021-01-04:
@fmauch: wouldn't the "the controller is not running" be the problem?

The error message is printed by the `joint_trajectory_controller` if I'm not mistaken.

fmauch, 2021-01-04:
> @fmauch: wouldn't the "the controller is not running" be the problem?
> 
> The error message is printed by the `joint_trajectory_controller` if I'm not mistaken.

Absolutely. However, by default the `scaled_pos_joint_traj_controller` is loaded (and started, if possible), while the `pos_joint_traj_controller` is [loaded but not started](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/launch/ur5e_bringup.launch#L9). Hence, my initial answer. 

To start the controller, however, the driver must be able to control the robot which requires the "External Control" URCap being installed and a program containing the "External Control" program node must be running as also explained in the quick start section. I am assuming that there has gone something wrong on that side. @hustfzt2 if you could post the output (in a code block) of the `roslaunch ur_robot_driver ur5e_bringup.launch` command that might help debugging your problem and support / contradict my assumption.

hustfzt2, 2021-01-07:
I have solved the issues,  when ROS driver is used to control the robot, the  "External Control"  in the teaching pendent must be firstly running by run button. thanks @gavanderhoorn @fmauch.

fmauch, 2021-05-31:
This looks resolved to me -> closing.

closed 2021-05-31 by fmauch