UniversalRobots/Universal_Robots_ROS_Driver #305: UR5 CB3 not moving when using /scaled_pos_joint_traj_controller action

FrederikPlahl, 2021-01-05:
# Summary
My UR5 CB3 is not moving but accepting the goals and setting them to active and also giving feedback when using the /scaled_pos_joint_traj_controller/follow_joint_trajectory action.

*Introduction to the issue*
I'm using the /scaled_pos_joint_traj_controller/follow_joint_trajectory action and sending 3 waypoints via a client I have written. Basically the shoulder_pan_joint should turn 90° and turn back into home position.
When sending the goal to the action I can see the /scaled_pos_joint_traj_controller/follow_joint_trajectory/feedback topic giving me the desired, actual and error positions. These positions are basically all the same (error is about 2,5 e-05) so I'd expect the robot to go to the next waypoint but it isn't. I also have goal tolerances for every joint of 0.1 so it should accept the goal.

I have no idea why it is not moving because also no error is posted anywhere.
PS: Speed slider is at 100% and program isn't paused.

## /scaled_pos_joint_traj_controller/follow_joint_trajectory/goal topic

`neobotix@mmo-500-1:~$ rostopic echo /scaled_pos_joint_traj_controller/follow_joint_trajectory/goal 
header: 
  seq: 0
  stamp: 
    secs: 1609853735
    nsecs: 760345310
  frame_id: ''
goal_id: 
  stamp: 
    secs: 1609853735
    nsecs: 760345861
  id: "/follow_joint_client-1-1609853735.760345861"
goal: 
  trajectory: 
    header: 
      seq: 0
      stamp: 
        secs: 1609853737
        nsecs: 760325830
      frame_id: ''
    joint_names: [shoulder_pan_joint, shoulder_lift_joint, elbow_joint, wrist_1_joint, wrist_2_joint,
  wrist_3_joint]
    points: 
      - 
        positions: [0.0, -1.5707963267948966, 0.0, -1.5707963267948966, 0.0, 0.0]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        effort: []
        time_from_start: 
          secs: 0
          nsecs:         0
      - 
        positions: [1.5707963267948966, -1.5707963267948966, 0.0, -1.5707963267948966, 0.0, 0.0]
        velocities: [1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
        accelerations: [1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
        effort: []
        time_from_start: 
          secs: 4
          nsecs:         0
      - 
        positions: [0.0, -1.5707963267948966, 0.0, -1.5707963267948966, 0.0, 0.0]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: [1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
        effort: []
        time_from_start: 
          secs: 7
          nsecs:         0
  path_tolerance: []
  goal_tolerance: 
    - 
      name: "shoulder_pan_joint"
      position: 0.1
      velocity: 0.1
      acceleration: 0.1
    - 
      name: "shoulder_lift_joint"
      position: 0.1
      velocity: 0.1
[... 231 lines cut ...]
  goal.goal_tolerance[4].acceleration = 0.1;

  goal.goal_tolerance[5].name = "wrist_3_joint";
  goal.goal_tolerance[5].position = 0.1;
  goal.goal_tolerance[5].velocity = 0.1;
  goal.goal_tolerance[5].acceleration = 0.1;

  goal.goal_time_tolerance = ros::Duration(2.0);

  //sending goal:
  client->sendGoal(goal);

  //result:
 
  while (client->getState() != actionlib::SimpleClientGoalState::SUCCEEDED && ros::ok()){
    client->waitForResult(ros::Duration(1)); 
    
    printf("Current State: %s\n", client->getState().toString().c_str());
    sleep(2);
  }
  printf("Action ended!\n");
  return 0;
};`

Thanks a lot in advance for helping!

fmauch, 2021-01-05:
We had a mail conversation on this earlier where I suggested, that the program might be paused, which @FrederikPlahl ruled out in the OP. Missing in the OP is the fact, that the driver's output indeed showed `Robot ready to receive control commands` as latest status update.

To be honest, I don't really have a clue what is going on here. 

You could show us what the control panel says on the variables tab. It should look either like this if you are using teh "Run program" main menu entry point:
[image]

Or like this if you use the "Program Robot" main menu entry point.
[image]

gavanderhoorn, 2021-01-05:
I would suggest to see whether a MoveIt generated trajectory results in motion.

That would give us something to compare.

FrederikPlahl, 2021-01-06:
Thanks for the fast reply! 
I now connected another PC to the UR5 to see if there is a problem with my PC but the control panel shows as it did with the other PC only the Program is starting... Please wait... message.

[image]

While the terminal still says "Robot ready to receive control commands"

[image]

So indeed it is kinda paused even though I started it. But I don´t know what is going on there.

gavanderhoorn, 2021-01-06:
Have you checked for a firewall running on any of the machines?

IP addresses correctly configured?

FrederikPlahl, 2021-01-06:
Firewalls are not active on my PC and as far as I know, there is no firewall on the UR. 

I'm not an expert in IP configuration so there could definitely be a problem. But the PC and the UR can communicate in the sense of I can ping the IP from the UR with positive feedback.
PC´s IP: 10.166.32.166 Subnet: 255.255.240.0
UR´s IP: 10.166.32.167 Subnet: 255.255.240.0

In the installation tab of the UR I have for the External Control configuration:

Host IP: 10.166.32.166
Custom port: 50001
Host name: 10.166.32.166

gavanderhoorn, 2021-01-06:
> Firewalls are not active on my PC 

how did you check?

> But the PC and the UR can communicate in the sense of I can ping the IP from the UR with positive feedback.

the symptom you describe (program hanging while trying to start it on the TP) is very often caused by either incorrect configuration of the external control urcap, or blocked ports on the PC side.

Being able to ping tells us ICMP is working, but not whether some port is being blocked or is unreachable (for whatever reason).

Why do you have `50001` for the port, instead of the default `50002`?

FrederikPlahl, 2021-01-06:
> how did you check?

I checked with ´´´sudo ufw status´´´ wich gave me an ´´´inactive´´´

> the symptom you describe (program hanging while trying to start it on the TP) is very often caused by either incorrect configuration of the external control urcap, or blocked ports on the PC side.

makes sense to me! 

> Why do you have 50001 for the port, instead of the default 50002?

I have no idea! Maybe I thought 50002 was the default but I will check with the default port again. Hopefully that will help

FrederikPlahl, 2021-01-06:
It was this port that was wrong! Thanks very much for helping!! 
Weirdly when I restart the robot the port is set back to 50001 and the IP also sets back to another IP but when I change both the robot moves.

No I need to check something in my code cause it is moving in a completly different direction as I expected...

closed 2021-01-06 by FrederikPlahl