UniversalRobots/Universal_Robots_ROS_Driver #306: Reading has_realtime fails sometimes

axelschroth, 2021-01-13:
I think this line should be changed from

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/783b1d1f5774220058d0a2aa3dd93185641e8c7e/ur_robot_driver/src/hardware_interface_node.cpp#L60

to

`  bool has_realtime = false;`

I noticed that if I modify some unrelated code without initializing `has_realtime`, printing `has_realtime` sometimes outputs a number like e.g. 168 or 66. I suspect the ifstream to do nothing if `/sys/kernel/realtime` does not exist.
Any thoughts on that strange behavior?

fmauch, 2021-01-14:
Using the stream's `fail()` or `good()` functions might be reasonable. I've wanted to refactor this code for quite some time to also allow setting a higher thread priority if no realtime capability is found.

Initializing the variable and checking the ifstream might be a good first start to address the issue you mentioned.

Something like 
```c++
std::ifstream realtime_file("/sys/kernel/realtime", std::ios::in);
bool has_realtime = false;
if (realtime_file.good())
{
  realtime_file >> has_realtime;
}

@axelschroth could you maybe test whether this works for you?

axelschroth, 2021-01-19:
Following works for me using `is_open()`:

```
std::ifstream realtime_file("/sys/kernel/realtime", std::ios::in);
bool has_realtime = false;
if (realtime_file.is_open())
{
  realtime_file >> has_realtime;
}
```

axelschroth, 2021-01-19:
Sorry for the confusion above. I created a PR.

fmauch, 2021-05-31:
This should have been fixed with #311 -> closing this issue.

closed 2021-05-31 by fmauch