UniversalRobots/Universal_Robots_ROS_Driver #307: Driver crash when loading controller

ManciuCen, 2021-01-13:
# Summary

When I using this driver and running to step "Loading controller : speed_scaling_state_controller", console will throw traceback message and then exit unnormally.

# Versions
 - ROS Driver version: melodic
 - Affected Robot Software Version(s): polyscope3.13
 - Affected Robot Hardware Version(s): UR5 CB3

# Issue details
When I run the driver and try to load two controllers, output messages as below:
[image]
If I run in debug mode, Here is the output message:
[image]
If I remove the speed_scaling_state_controller, it seems everythings normal. But when I use ctrl+c to exit, it still warn me that "Controller Spawner error while taking down controllers", as picture showes below:
[image]

I use the same code test on another computer, and everythings good. They are both 18.04 bionic and ROS melodic.
Actually, I also compare the controller_manager, controller_manager_msgs, etc. package version, and they are the same(newest).
So I was confused about what makes the running environment different and cause such error crash, and how can I fix it?
Many thanks!

gavanderhoorn, 2021-01-13:
Something to check: do you have any outstanding updates to ROS packages? If so: update all ROS packages, remove the `build` and `devel` folder from your workspace (run `catkin clean -y` if using `catkin_tools`) and rebuilt everything.

There is no ABI compatibility guarantee in ROS, so rebuilding packages in your workspace is a requirement after updating any packages provided by the ROS buildfarm.

I'm not saying this is the cause here, just something to verify.

`SEGAULT`s can be caused by ABI problems, especially when working with plugins (which controllers are).

ManciuCen, 2021-01-18:
@gavanderhoorn Thanks! It did works!

closed 2021-01-18 by ManciuCen

gavanderhoorn, 2021-01-18:
Not sure whether it was really the cause in this case, but problems with ABI compatibility may be something to add to the FAQ section @fmauch ?

fmauch, 2021-01-18:
> Not sure whether it was really the cause in this case, but problems with ABI compatibility may be something to add to the FAQ section @fmauch ?

That sounds like a good idea. I'm a little short on time currently, so I'll add a new issue for this.