UniversalRobots/Universal_Robots_ROS_Driver #308: MoveIt and UR Driver: "Can't accept new action goals. Controller is not running."

Schellenberg3, 2021-01-13:
# Summary

While trying to use MoveIt! to plan and execute a trajectory, the UR driver raises an error expressing that it `Can't accept new action goals. Controller is not running`. 

However, the `External Control` script is being run (on the run menu, not through the program menu) and the terminal displays `Robot ready to receive control commands.`

To resolve some other errors, I've made the following changes:

1) Adjust the  file `ur5_e_moveit_config/config/controllers.yaml` so that the name is ` /pos_joint_traj_controller`. I followed advice from [issue 261](https://github.com/ros-industrial/ur_modern_driver/issues/261). See file below:
```
controller_list:
 - name: /pos_joint_traj_controller # or /vel_based_pos_traj_controller or /pos_based_pos_traj_controller
   action_ns: follow_joint_trajectory
   type: FollowJointTrajectory
   default: true
   joints:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint](url)
```
2) Adjust the  file `ur5_e_moveit_config/launch/move_group.launch` so that  'MoveGroupExecuteService' is instead 'MoveGroupExecuteTrajectoryAction'. Followed [issue 416](https://github.com/ros-industrial/universal_robot/issues/413). See file snippet below:
```
    <!-- MoveGroup capabilities to load -->
    <param name="capabilities" value="move_group/MoveGroupCartesianPathService
				      move_group/MoveGroupExecuteTrajectoryAction
				      move_group/MoveGroupKinematicsService
				      move_group/MoveGroupMoveAction
				      move_group/MoveGroupPickPlaceAction
				      move_group/MoveGroupPlanService
				      move_group/MoveGroupQueryPlannersService
				      move_group/MoveGroupStateValidationService
				      move_group/MoveGroupGetPlanningSceneService
				      move_group/ClearOctomapService
				      " />
```
# Versions
 - ROS Driver version: Melodic on Ubuntu 18.04
 - Affected Robot Software Version(s): 5.9.3.1031212
 - Affected Robot: UR5e
 - URCaps Software version(s): External Control 1.0.4

# Issue details
*This part is important in order to clarify the severity of the issue.*
I am running 3 terminal windows related to this. Their commands are:

1)`roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=192.168.0.101`
2)`roslaunch ur5_e_moveit_config ur5_e_moveit_planning_execution.launch `
3)`roslaunch ur5_e_moveit_config moveit_rviz.launch config:=true`

My process has been as follows:
1) Run terminal 1
2) Start the external control program on `Run` tab of the teach pendant
3) Run terminal 2
4) Run terminal 3
5) Use the RVIZ GUI to drag the robot to a new location
6) Press `plan` in the RVIZ GUI
7) Press `execute` in the RVIZ GUI

At this point the robot doesn't move and the `Can't accept new action goals. Controller is not running.` error appears in terminal 1.

I am fairly inexperienced with ROS and I think I've reached the limit of what I can solve on my own or by looking at other peoples issues. Any help to get this robot moving would be appreciate!

I've recorded the output of each terminal below, with notes to identify what is going on.

## Terminal 1: ur_robot_driver
*What does the setup look like and what are the objectives?*

[... 576 lines cut ...]
[ INFO] [1610567436.126810284]: Stereo is NOT SUPPORTED
[ INFO] [1610567436.126888819]: OpenGl version: 3.1 (GLSL 1.4).
[ INFO] [1610567439.343985600]: Loading robot model 'ur5e'...
[ WARN] [1610567439.344022688]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1610567439.344041725]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1610567439.357535066]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/rviz_disl_Alienware_10832_3193070364466814833/manipulator/kinematics_solver_attempts' from your configuration.
[ INFO] [1610567439.420748867]: Starting planning scene monitor
[ INFO] [1610567439.422401635]: Listening to '/move_group/monitored_planning_scene'
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
[ INFO] [1610567439.646497431]: Constructing new MoveGroup connection for group 'manipulator' in namespace ''
[ INFO] [1610567440.696956330]: Ready to take commands for planning group manipulator.
[ INFO] [1610567440.697010611]: Looking around: no
[ INFO] [1610567440.697032565]: Replanning: no
[ INFO] [1610567440.728937591]: Marker 'EE:goal_ee_link/10': mesh_use_embedded_materials is ignored.
```
The robot appears in the GUI (matches the physical position of the robot) and a trajectory to the new state is planned.
[image]
The following appears when `execute` is pressed:
```
[ INFO] [1610567749.578187378]: ABORTED: Solution found but controller failed during execution
```

gavanderhoorn, 2021-01-13:
As you can see in the log:

```
...
[INFO] [1610567071.604805]: Controller Spawner: Loaded controllers: pos_joint_traj_controller, joint_group_vel_controller
[INFO] [1610567071.606776]: Loading controller: force_torque_sensor_controller
[INFO] [1610567071.610747]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[INFO] [1610567071.612684]: Started controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
...
```

the following controllers are only loaded: `pos_joint_traj_controller, joint_group_vel_controller`, while these controllers are loaded *and started*: `joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller`.

So you have two options:

 1. start the `pos_joint_traj_controller`, or (and this would be better)
 1. change the `name` in the `controller_list` of your MoveIt config to `scaled_pos_joint_traj_controller`

in general you'll want to use the *scaled* versions of all these controllers (if it exists) as those controllers will take the state of the speed slider (on the TP) and the automatic internal scaling of the UR controller into account.

Schellenberg3, 2021-01-14:
Thank you @gavanderhoorn, this seems to have been my issue! With `scaled_pos_joint_traj_controller` we were able to get the robot to move.

closed 2021-01-14 by Schellenberg3

Aishwaryak2998, 2023-02-02:
@gavanderhoorn Hey, I seem to have the same problem you pointed out. Two of the controllers are only loaded and not loaded+started. I checked the name in the controller_list and it is scaled_pos_joint_traj_controller. The problem still unfortunately persists.

k-ish0212, 2023-03-30:
Im getting same issue..

Rafael00Soto, 2023-06-29:
> As you can see in the log:
> 
> ```
> ...
> [INFO] [1610567071.604805]: Controller Spawner: Loaded controllers: pos_joint_traj_controller, joint_group_vel_controller
> [INFO] [1610567071.606776]: Loading controller: force_torque_sensor_controller
> [INFO] [1610567071.610747]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
> [INFO] [1610567071.612684]: Started controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
> ...
> ```
> 
> the following controllers are only loaded: `pos_joint_traj_controller, joint_group_vel_controller`, while these controllers are loaded _and started_: `joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller`.
> 
> So you have two options:
> 
>     1. start the `pos_joint_traj_controller`, or (and this would be better)
> 
>     2. change the `name` in the `controller_list` of your MoveIt config to `scaled_pos_joint_traj_controller`
> 
> 
> in general you'll want to use the _scaled_ versions of all these controllers (if it exists) as those controllers will take the state of the speed slider (on the TP) and the automatic internal scaling of the UR controller into account.

Hi, where exactly is this controller_list? Is it the yaml file in /ur5e_moveit_config/config

puznik, 2024-08-24:
This solved it for me:
https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/4
Needed to press play on the robot tablet to actually run the program.