UniversalRobots/Universal_Robots_ROS_Driver #312: Connection to robot drops after starting the program in the robot

failiz, 2021-01-21:
# Summary

I did not manage to make the robot move. The driver connects to the robot after the program is run, but disconnects a few milliseconds later.  

# Versions
 - ROS Driver version: melodic
 - Affected Robot Software Version(s):  3.14.3.1031232
 - Affected Robot Hardware Version(s): UR5 (CB3UR5)
 - Robot Serial Number: 2015350824
 - UR+ product(s) installed: None
 - URCaps Software version(s): 1.0.4

# Impact
I cannot move the robot

# Issue details
1. I launch the driver.
2. After that, I run the program in the robot.
3. When I launch the "rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller" sometimes the controller does not show up. When I can select the controller "scaled_pos_joint_traj_controller", the sliders apper but moving them does not move the robot

I have tried to:
1. I launch the driver.
2. Run the program in the robot with: rosservice call /ur_hardware_interface/dashboard/play
3. And again, does not move with "rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller" 

```
roslaunch ur_robot_driver ur5_bringup.launch robot_ip:=192.168.56.10 kinematics_config:=/mnt/c/fai/documents/git/ROS_UR_driver/my_robot_calibration.yaml
... logging to /home/fai/.ros/log/a0cdd1b2-5bf1-11eb-9ca9-00090ffe0001/roslaunch-PC622106-31973.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://PC622106:52046/

SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 125
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /hardware_control_loop/loop_hz: 125
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 125
 * /joint_state_controller/type: joint_state_contr...
 * /pos_joint_traj_controller/action_monitor_rate: 20
 * /pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/goal_time: 0.6
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_joint_traj_controller/state_publish_rate: 125
 * /pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /pos_joint_traj_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /robot_status_controller/handle_name: industrial_robot_...
 * /robot_status_controller/publish_rate: 10
 * /robot_status_controller/type: industrial_robot_...
[... 227 lines cut ...]
[INFO] [1611237898.819480]: Controller Spawner: Loaded controllers: pos_joint_traj_controller, joint_group_vel_controller
[INFO] [1611237898.887216]: Loading controller: force_torque_sensor_controller
[INFO] [1611237899.002197]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[INFO] [1611237899.103373]: Started controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[ INFO] [1611237912.553321300]: Robot requested program
[ INFO] [1611237912.553631200]: Sent program to robot
[ INFO] [1611237912.612828300]: Robot ready to receive control commands.
[ INFO] [1611237912.663855800]: Connection to robot dropped, waiting for new connection.
[ERROR] [1611237912.767024500]: Could not stop requested controllers

`````

The log in the robot can be seen [here](https://www.dropbox.com/s/udnlxc4yizf34b4/11.jpg?dl=0). 

## Project status at point of discovered
*When did you first observe the issue?*
When trying to setup the ROS driver.

## Expected Behavior
Robot moving

## Actual Behavior
Robot does not move. There are errors in the driver output. "[ERROR] [1611237897.488163500]: Could not get fresh data package from robot" appears sometimes. Even if they do not appear, I cannot move the robot.

"[ERROR] [1611237912.767024500]: Could not stop requested controllers" appears sometimes too.  Even if they do not appear, I cannot move the robot.

fmauch, 2021-01-22:
What is your connection between your ROS pc and your robot? This looks like there's an unreliable connection like a WIFI connection active.

failiz, 2021-01-26:
I am using an Ethernet cable but with an USB-C to Ethernet adaptor. I will try with another computer with an integrated Ethernet card. Have you seen that these adaptors cause issues before?

gavanderhoorn, 2021-01-26:
There are reports of connection issues by users who use USB adapters, yes.

See #182.

That hasn't been completely resolved though, so whether it's the cause is unknown at this point.

stefanscherzinger, 2021-09-01:
@failiz Any updates on this?

failiz, 2021-09-06:
Hi,
Sorry for the delayed response, I had limited access to the robot.
I managed to install it and move the robot in a computer with an integrated Ethernet card (not using the USB-C to Ethernet adapter). So, I guess we can close the issue. 
Thanks for the support!

closed 2021-09-06 by failiz