UniversalRobots/Universal_Robots_ROS_Driver #313: Unable to start External Control URCap

waiyc, 2021-01-26:
# Summary

I was able to established connection via Ethernet cable from the UR3e to my laptop for a while. Out of a sudden I encountered this External Control URCap issue. 

External Control URcap stucked at 'Starting Program' pop-up window on teach pendant. The pop-up window will turn into a error window once I kill the bring-up rosnode on the laptop. 

```
Compile error: An expression was found which produces a result that is not used in the program

Program node : Control by xx.xxxx.xx.xx 
Program node type : External Control
Error position : line 2
URCap : External Control 
               FZI Research center for Information Technology

Script Code
Error position: line 37 column 2 
...
    #begin: URCap Program Node
    #       Source : External Control, 1.0.4, FZI Research Center for Information ... 
    #       Type: External Control 
    $ 2 "Control by xx.xxx.xx.xx"
null        #end : UPCap Program Node
    end 
end
```

# Versions
 - ROS Driver version: Melodic
 - Affected Robot Software Version(s): URSoftware 5.9.2.1031163(Aug 31 2020)
 - Affected Robot Hardware Version(s): UR3e
 - Robot Serial Number: 20205300771
 - UR+ product(s) installed: None
 - URCaps Software version(s): External Control 1.0.4

ipa-rwu, 2021-02-05:
same issue, teach pendant stuck at  'Starting Program... Please wait...' as shown in this [picture](https://drive.google.com/file/d/1CcwgkNsTfapR7xVri-jOBYRsukPvxZMT/view?usp=sharing)

fmauch, 2021-02-05:
The driver is running on ROS side? There is no firewall stopping things from working? IP addresses are [configured correctly](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/install_urcap_e_series.md) on the teach pendant? What's the output on the ROS side?

waiyc, 2021-02-05:
`The driver is running on ROS side? `
Yes

`There is no firewall stopping things from working? `
How should I check this?

`IP addresses are configured correctly on the teach pendant? `
Yes

`What's the output on the ROS side?`
Robot is ready to control is shown on the terminal. 
But the 'Starting Program' pop-up window on teach pendant does not go off

fmauch, 2021-02-05:
Checking the status of ufw can be done with `sudo ufw status`. But, there might be a different firewall running, as this is only one way to use a firewall on ubuntu.

What is the output of `ss -atnp | grep 5000`? It should be similar to 
```
$ ss -atnp | grep 5000
LISTEN     0      1                                        0.0.0.0:50001                             0.0.0.0:*     users:(("ur_robot_driver",pid=1396538,fd=18))
LISTEN     0      1                                        0.0.0.0:50002                             0.0.0.0:*     users:(("ur_robot_driver",pid=1396538,fd=16))
```

It would be good to see the **complete** output of the driver, not only the most recent line.

waiyc, 2021-02-23:
Problem solves after reconfigure the IP address on both PC and robots

closed 2021-02-23 by waiyc