UniversalRobots/Universal_Robots_ROS_Driver #317: The result of ur_calibration may be not correct

jsbyysheng, 2021-02-08:
# Summary

Your issue may already be reported!
Please search on the [issue track](../) before creating one.

*Introduction to the issue*
The result of ur_calibration may be not correct.

# Versions
 - ROS Driver version: master (newest)
 - Affected Robot Software Version(s): 3.12.0
 - Affected Robot Hardware Version(s):
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s): 

# Impact
*What is the impact of the issue. Try to describe it well enough, in order for us to prioritise the issues.*

If I use this calibration file to bringup my ur5, rviz gives me a strange robot (calibration_devel, fmauch/universal_robot):

[image]

# Details

Recently, I want to extract calibration information by executing the following as descriped in the instrcution.
```bash
$ roslaunch ur_calibration calibration_correction.launch robot_ip:=<robot_ip> target_filename:="${HOME}/my_robot_calibration.yaml"
```

The terminal output is 

```
SUMMARY
========

PARAMETERS
 * /calibration_correction/output_filename: /home/robotmws/Wo...
 * /calibration_correction/robot_ip: 192.168.15.3
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    calibration_correction (ur_calibration/calibration_correction)

auto-starting new master
process[master]: started with pid [24917]
ROS_MASTER_URI=http://rosmaster:11311/

setting /run_id to 18245fa8-69f1-11eb-b6c7-e4e74962b568
process[rosout-1]: started with pid [24935]
started core service [/rosout]
process[calibration_correction-2]: started with pid [24942]
[DEBUG]: Starting up producer
[WARNING]: No realtime capabilities found. Consider using a realtime system for better performance
[WARNING]: Failed to read from stream, reconnecting in 1 seconds...
[DEBUG]: Invalid robot package type recieved: 24
[DEBUG]: Invalid robot package type recieved: 23
[DEBUG]: Invalid robot package type recieved: 5
[INFO]: checksum: [2052432846 1571393760 4068094558 1950685856 863700637 844303014 ]
dh_theta: [-2.9832203171837e-06 -0.288176963658702 -0.207180705671394 0.495349544718743 -6.21161769466526e-05 6.52377021472243e-05 ]
dh_a: [-4.80668983106896e-05 -0.233680645546451 -0.187647667768473 -0.000106752836465411 -5.27320231715568e-05 0 ]
dh_d: [0.151878269099865 -36.9284767740322 78.0679702897434 -41.0273539212836 0.0853111795260134 0.0823216894157751 ]
dh_alpha: [1.57125058393544 0.00187554662077913 -0.00246443022046067 1.57136678055224 -1.57151512729844 0 ]
calibration_status: 2

[DEBUG]: Destructing pipeline
[DEBUG]: Stopping pipeline! <Pipeline>
[DEBUG]: Pipeline consumer ended! <Pipeline>
[... 539 lines cut ...]
    rviz_HP_ZBook_17_G3_26432_3118122257842318432 (rviz/rviz)

ROS_MASTER_URI=http://rosmaster:11311/

process[rviz_HP_ZBook_17_G3_26432_3118122257842318432-1]: started with pid [26452]
[ INFO] [1612777770.800955319]: rviz version 1.13.15
[ INFO] [1612777770.800995477]: compiled against Qt version 5.9.5
[ INFO] [1612777770.801009131]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1612777770.804035440]: Forcing OpenGl version 0.
[ INFO] [1612777771.481776404]: Stereo is NOT SUPPORTED
[ INFO] [1612777771.481825749]: OpenGl version: 4.6 (GLSL 4.6).
[ INFO] [1612777774.796662368]: Loading robot model 'ur5'...
[ WARN] [1612777774.796707216]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1612777774.796732321]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1612777774.810429408]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/rviz_HP_ZBook_17_G3_26432_3118122257842318432/manipulator/kinematics_solver_attempts' from your configuration.
[ INFO] [1612777774.882449074]: Starting planning scene monitor
[ INFO] [1612777774.884021889]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1612777775.119095118]: Constructing new MoveGroup connection for group 'manipulator' in namespace ''
[ INFO] [1612777776.189463892]: Ready to take commands for planning group manipulator.
[ INFO] [1612777776.189544348]: Looking around: no
[ INFO] [1612777776.189605759]: Replanning: no
```

All work well if I do not use the calibration file.

fmauch, 2021-02-09:
Maybe a silly question - but are you sure that you are actually using a UR5?

The following image shows the model of a UR3 using your calibration file:
[image]

To verify, what is the output of 
```
rosservice call /ur_hardware_interface/dashboard/raw_request "query: 'get robot model'"
```

jsbyysheng, 2021-02-10:
Aha, sorry to interrupt you... I review many examples about UR5 and forget mine is UR3. It's a stupid problem indeed...

closed 2021-02-10 by jsbyysheng