UniversalRobots/Universal_Robots_ROS_Driver #323: Passing control back and forth between local pendant program and ROS

objimp, 2021-02-24:
I'm working on a shared control application where we would like to interrupt and take control of a TP program and then resume into the program.   I think we have a workaround by stopping the program , starting an external control program,  stopping that program, and restarting the first program over the dashboard interface, but saw in a comment on #162 , that 

"A mode where the controller can 'request' the ROS side to become active (which is planned as a future extension to the driver), with the program on the TP in full control, would perhaps be better."

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/162#issuecomment-617832629

Which is exactly what we are trying to do...

Is there any prognosis or timing for this future extension?  

We can proceed with the dashboard interface approach (which also involves some very awkward constructs to pick up a program where it left off), but if there is a better way more aligned with future features we would rather live in the future.

Thanks.

fmauch, 2021-03-01:
Currently, there is nothing being implemented in that direction.

The driver can hand back control to the TP (Which will effectively finish the program node) by calling the [`hand_back_control`](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#hand_back_control-std_srvstrigger) service.

We are currently working to enable having the "External Control" program node inside a program multiple times. See https://github.com/UniversalRobots/Universal_Robots_ExternalControl_URCap/pull/14 for details.

With this, for your setup you would still require a trigger to call the `hand_back_control` service, though.

cambel, 2021-04-01:
@fmauch I have a question regarding `hand_back_control`. I tested it but it does not completely release the control of the robot.
 Here is the result of calling `rosservice call /ur_hardware_interface/hand_back_control` 
> success: True
message: "Deactivated control"

However, the program still is alive
>[ INFO] [1617248241.990693264]: Connection to robot dropped, waiting for new connection.
[ INFO] [1617248241.991454780]: Robot ready to receive control commands.

It seems to immediately reconnect, is that how it should work?

fmauch, 2021-04-03:
That depends on the program, you created on the robot. If you have a program only containing the external_control node and the program loops forever (See checkbox in screenshot below, which I think is enabled by default), it will obviously go right back into control.

[image]

cambel, 2021-04-03:
@fmauch I see, thank you!

stefanscherzinger, 2021-08-09:
Now, this should be doable with the [library's support for various URCap nodes](https://github.com/UniversalRobots/Universal_Robots_ExternalControl_URCap#multiple-urcap-nodes). 

> With this, for your setup you would still require a trigger to call the hand_back_control service, though.

And this part could be triggered from within the new [ROS as a service URCap](https://github.com/UniversalRobots/Universal_Robots_ROS_as_a_Service_URCap/blob/master/doc/tutorial.md#call-a-service).

closed 2021-08-09 by stefanscherzinger