UniversalRobots/Universal_Robots_ROS_Driver #324: UR simulator in docker on windows not working

BlazPotokar, 2021-02-26:
I am trying to create and run docker container of ur simulator. It work well on linux host, but not on windows. I use docker instructions provided on this repo inside `.github\dockerursim` folder.
Because I am unable to run originally provided `build_and_run_docker_ursim.sh` in Cmder emulator, I manually do those steps. 

```
docker network create --subnet=192.168.0.0/16 static_test_net
docker volume create dockerursim

docker build ./Dockerfile -t mydockerursim

docker run --name="mydockerursim" -d \
  -e ROBOT_MODEL=UR5 \
  --net static_test_net \
  --ip 192.168.56.101 \
  -p 8080:8080 \
  -p 29999:29999 \
  -p 30001-30004:30001-30004 \
  -v "c:/.vol":/ursim/programs \
  -v dockursim:/ursim \
  --privileged \
  --cpus=1 \
  mydockerursim
```
When I run http://localhost:8080/ in browser I do not get GUI (but it works on linux). It is only black screen with mouse. At first moment I get popup for a blink of a second (Connecting...). When I stop container I do get another popup "connection error". So gui partually works. It only shows this two pupups but background is black. One try it did actually showed gui normally somehow, but I cannot redo it anymore.

[image]

fmauch, 2021-03-01:
I am afraid, this goes beyond the scope of this repository.

The docker configuration used for continuous integration is not intended to be used on personal machines. If you want to run your own simulation locally, best use the VirtualBox images provided on the UR download page.

BlazPotokar, 2021-03-01:
Hi, I managed to partially solve problem.  There is no problem with graphics itself, but starting program. I think init script somehow fails at execution. But it also does not finish and is still running because, otherwise docker container should shutdown, if I understand it correctly. 
My solution is to trigger `run` script manually or in my case, I added command in my docker-compose file like this:
`/bin/execlineb /etc/services.d/ursim/run  UR10`
I am not sure if this command launches program properly, but so far I didn't notice any problems.

closed 2021-03-05 by BlazPotokar

BlazPotokar, 2021-03-11:
nbhgj