UniversalRobots/Universal_Robots_ROS_Driver #326: Unable to control UR3 hardware using ROS Noetic

Haoran-Zhao, 2021-03-01:
# Summary
*Introduction to the issue*
Dear members of this repo's community,
I downloaded and built the unversal_robot_ros_driver package; followed the setup process in Readme.md and extracted calibration information; create external control program on UR3 TP, and pressed play button. However, when I tried "rosrun rqt_joint_trajectory_controller rqt_joint_trajectrory_controller", the drop down list is empty, I couldn't find joint slide bars.

Also after I execute the quick start command and press the play button of the external control program. I still didn't see `Robot ready to receive control commands. `
I also verify it may not be a issue of firewall. I create a serve on the ROS pc with command `nc -l -p 50002`, and create a client from UR3 linux system with command `nc 192.168.1.7 50002`. They both can show same output.

# Versions
 - ROS Driver version: ROS Noetic, ubuntu 20.04
 - Affected Robot Software Version(s): 3.17
 - Affected Robot Hardware Version(s): UR3
 - Robot Serial Number: 
 - UR+ product(s) installed:
 - URCaps Software version(s): external control-1.0.4.urcap

# Impact
Universal_robots_ROS_driver compatibility with Noetic and Ubuntu 20.04

## Steps to Reproduce

1. run following command on ROS pc
`roslaunch ur_robot_driver ur3_bringup.launch robot_ip:=192.168.1.5`
`
NODES
  /
    controller_stopper (controller_stopper/node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ros_control_controller_spawner (controller_manager/spawner)
    ros_control_stopped_spawner (controller_manager/spawner)
    ur_hardware_interface (ur_robot_driver/ur_robot_driver_node)
  /ur_hardware_interface/
    ur_robot_state_helper (ur_robot_driver/robot_state_helper)

auto-starting new master
process[master]: started with pid [468107]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3ccdf4a2-7aba-11eb-af7a-115ed6dc3b5b
process[rosout-1]: started with pid [468117]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [468124]
process[ur_hardware_interface-3]: started with pid [468125]
process[ros_control_controller_spawner-4]: started with pid [468126]
process[ros_control_stopped_spawner-5]: started with pid [468127]
process[controller_stopper-6]: started with pid [468128]
process[ur_hardware_interface/ur_robot_state_helper-7]: started with pid [468129]
[ INFO] [1614622587.882746522]: Waiting for controller manager service to come up on /controller_manager/switch_controller
[ INFO] [1614622587.890756554]: Initializing dashboard client
[ INFO] [1614622587.902132090]: waitForService: Service [/controller_manager/switch_controller] has not been advertised, waiting...
[ INFO] [1614622587.907281407]: waitForService: Service [/ur_hardware_interface/dashboard/play] has not been advertised, waiting...
[ INFO] [1614622587.929878232]: Initializing urdriver
Warning: No realtime capabilities found. Consider using a realtime system for better performance
         at line 396 in /home/cougarnet.uh.edu/hzhao9/Universal_Robots_Client_Library/include/ur_client_library/comm/pipeline.h
[ INFO] [1614622587.932561322]: waitForService: Service [/ur_hardware_interface/dashboard/play] is now available.
[INFO] [1614622588.096099]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1614622588.111261]: Controller Spawner: Waiting for service controller_manager/load_controller
Error:   The calibration parameters of the connected robot don't match the ones from the given kinematics config file. Please be aware that this can lead to critical inaccuracies of tcp positions. Use the ur_calibration tool to extract the correct calibration from the robot and pass that into the description. See [https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#extract-calibration-information] for details.
         at line 45 in /home/cougarnet.uh.edu/hzhao9/Universal_Robots_Client_Library/src/ur/calibration_checker.cpp
Warning: No realtime capabilities found. Consider using a realtime system for better performance
         at line 396 in /home/cougarnet.uh.edu/hzhao9/Universal_Robots_Client_Library/include/ur_client_library/comm/pipeline.h
Warning: No realtime capabilities found. Consider using a realtime system for better performance
         at line 396 in /home/cougarnet.uh.edu/hzhao9/Universal_Robots_Client_Library/include/ur_client_library/comm/pipeline.h
[ INFO] [1614622590.064833996]: Loaded ur_robot_driver hardware_interface
[ INFO] [1614622590.112331475]: waitForService: Service [/controller_manager/switch_controller] is now available.
[ INFO] [1614622590.112399044]: Service available.
[ INFO] [1614622590.112438795]: Waiting for controller list service to come up on /controller_manager/list_controllers
[ INFO] [1614622590.113733415]: Service available.
[INFO] [1614622590.215683]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1614622590.220736]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1614622590.225574]: Loading controller: joint_state_controller
[INFO] [1614622590.231601]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1614622590.239752]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1614622590.244222]: Loading controller: scaled_pos_joint_traj_controller
[INFO] [1614622590.245066]: Loading controller: pos_joint_traj_controller
[ INFO] [1614622590.294420019]: Robot mode is now RUNNING
[ INFO] [1614622590.294498128]: Robot's safety mode is now NORMAL
[INFO] [1614622590.308167]: Loading controller: speed_scaling_state_controller
[INFO] [1614622590.356234]: Loading controller: joint_group_vel_controller
[INFO] [1614622590.364227]: Loading controller: force_torque_sensor_controller
[INFO] [1614622590.380164]: Controller Spawner: Loaded controllers: pos_joint_traj_controller, joint_group_vel_controller
[INFO] [1614622590.388053]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[INFO] [1614622590.396364]: Started controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[ INFO] [1614622598.455227905]: Robot requested program
[ INFO] [1614622598.455376943]: Sent program to robot
`

2.run
`rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller`
empty drop-down list.
## Expected Behavior
*What did you expect and why?*
1. control the UR3 with joint slider.
2. control the UR3 with RVIZ.
3. see `Robot ready to receive control commands. ` after press the play button of the external control program

## Actual Behavior
*What did you observe? If possible please attach relevant information.*
1. can not control the UR3 with joint slider.
2. can plan the trajectory but can not execute the plan and control the UR3 with RVIZ.
3. There is only
`
[ INFO] [1614622598.455227905]: Robot requested program
[ INFO] [1614622598.455376943]: Sent program to robot
`
no
`
[ INFO] [1614622598.455376943]: Robot ready to receive control commands.
`
After press the play button of the external control program.

fmauch, 2021-03-02:
OK, if you see the `Sent program to robot` output, it would be interesting to see what is actually happening on the teach pendant.

Does the program keep running? Does the program finish? Is there an error popping up?

The driver will print `Robot ready to receive control commands.` when it receives the keepalive signals from the script sent [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/resources/ros_control.urscript#L106). For this to work, the driver opens a socket on port `50001` (by default). Verify that this port is reachable from the ur controller by using 

```
nc 192.168.1.7 50001
```

You should see the output `Robot ready to receive control commands` immediately and `Connection to robot dropped, waiting for new connection.` as soon as you stop the `nc` command. If the above is not working, there is most likely some firewall stopping communication. If that works, but the driver doesn't, we'll have to look at the generated script code. For this, see [this comment](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/74#issuecomment-571949829).

Haoran-Zhao, 2021-03-02:
> OK, if you see the `Sent program to robot` output, it would be interesting to see what is actually happening on the teach pendant.
>
> Does the program keep running? Does the program finish? Is there [...]

@fmauch Thanks so much for your reply. After I saw `Sent program to robot`, the TP is still running and no error popped up. The default port seems to be 50002 of my UR3. I tried to switch to 50001, and the "connection time out" error on TP popped out. I guess it's because of the firewall, so I have contacted the IT department to open the port for me and I will do the test and update. 

To verify if the port is reachable or fully usable, are the `nc -l -p 50001` on ROS PC and `nc 192.168.1.7 50001` on TP sufficient?

fmauch, 2021-03-02:
You will need both, port 50002 and 50001. 50002 is responsible for requesting the UrScript from the driver, while 50001 handles the keepalive signal. Therefore, the port configured on the teach pendant should be 50002 if not changed on the ROS side.

> To verify if the port is reachable or fully usable, are the nc -l -p 50001 on ROS PC and nc 192.168.1.7 50001 on TP sufficient?

Probably yes, but if you run the driver on the ROS pc both ports should be opened by the driver directly. No need for the `nc -l` command in this case.

Haoran-Zhao, 2021-03-02:
@fmauch Thanks so much, the reason for no `Robot ready to receive control commands` is because the port 50001 is blocked.
Now I can control the robot with `rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller`.

However, when I try to use moveit and control the arm using RVIZ, it can plan the trajectory but can not execute the plan, and give me the following error:
`
[ INFO] [1614724359.357656721]: ParallelPlan::solve(): Solution found by one or more threads in 0.002595 seconds
[ INFO] [1614724359.357937732]: manipulator/manipulator: Starting planning with 1 states already in datastructure
[ INFO] [1614724359.358142362]: manipulator/manipulator: Starting planning with 1 states already in datastructure
[ INFO] [1614724359.359437043]: manipulator/manipulator: Created 4 states (2 start + 2 goal)
[ INFO] [1614724359.360046119]: manipulator/manipulator: Created 4 states (2 start + 2 goal)
[ INFO] [1614724359.360238626]: ParallelPlan::solve(): Solution found by one or more threads in 0.002450 seconds
[ INFO] [1614724359.368391193]: SimpleSetup: Path simplification took 0.007968 seconds and changed from 3 to 2 states
[ INFO] [1614724360.660168262]: Execution request received
[ERROR] [1614724360.670311496]: Can't accept new action goals. Controller is not running.
[ WARN] [1614724360.671032015]: Controller  failed with error INVALID_GOAL: 
[ WARN] [1614724360.671237323]: Controller handle  reports status FAILED
[ INFO] [1614724360.671357665]: Completed trajectory execution with status FAILED ...
[ INFO] [1614724360.671559482]: Execution completed: FAILED
[ INFO] [1614724360.698728620]: ABORTED: Solution found but controller failed during execution
`
I have changed the `action_ns: follow_joint_trajectory` to `action_ns: /scaled_pos_joint_traj_controller/follow_joint_trajectory` in the controllers.yaml file. Do you have any clue about this problem? Thanks a lot.

Haoran-Zhao, 2021-03-03:
@fmauch Many thanks for your help, I finally figure it out that why the controller is not started. Actually it's the problem of the TP side. While I was using melodic, I don't need to click the Stop (square button)  before I start my program. Pause and play button worked fine. But for the Noetic, I should click stop button first, then run my code then click play button.
And it's also neccesary to modify the name space as instruct in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/308#issue-785405743

closed 2021-03-03 by Haoran-Zhao