UniversalRobots/Universal_Robots_ROS_Driver #328: Possible to change speed scaling value from ROS?

DanielSchimpf, 2021-03-03:
# Summary

ROS Kinetic
Ubuntu 16
UR 10e

Is it possible to change the speed scaling value [0-1] from ROS (similar to how IO's are set)? I would like a C++ node on my computer to change the speed scaling value. Or can it only be set from the teaching panel? The reason why I am asking is that I want to change the speed of a trajectory during the execution.

fmauch, 2021-03-07:
Would [this service](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#set_speed_slider-ur_msgssetspeedsliderfraction) be of any help?

DanielSchimpf, 2021-03-08:
Must have missed it when going through the list. "rosservice call /ur_hardware_interface/set_speed_slider 0.5" changes the speed scaling value as expected. 

Thank you @fmauch!

closed 2021-03-08 by DanielSchimpf