UniversalRobots/Universal_Robots_ROS_Driver #332: AttributeError: 'SetModeActionGoal' object has no attribute 'target_robot_mode'

dionysius07, 2021-03-10:
# Summary

I am attempting to restart the External Control program on the teach pendant via a Python script to test restarts in the event of a program stop. According to the documentation, it is recommended to use the 'set_mode' action server, which is what I have been trying to do, however whenever I try sending the SetModeActionGoal to the action server, I get the error in the title:

# Versions
 - ROS Driver version: ROS Noetic
 - Affected Robot Software Version(s): URSoftware 3.11.0.82155
 - Affected Robot Hardware Version(s): UR5-CB3
 - URCaps Software version(s): External Control 1.0.4

# Issue details
This is the traceback of the error:
```
Traceback (most recent call last):
  File "/home/dennis/Workspaces/binpick_ws/src/binpick_universal_robot/binpick_ur_moveit/src/actiontest.py", line 15, in <module>
    action_client.send_goal(message)
  File "/opt/ros/noetic/lib/python3/dist-packages/actionlib/simple_action_client.py", line 92, in send_goal
    self.gh = self.action_client.send_goal(goal, self._handle_transition, self._handle_feedback)
  File "/opt/ros/noetic/lib/python3/dist-packages/actionlib/action_client.py", line 561, in send_goal
    return self.manager.init_goal(goal, transition_cb, feedback_cb)
  File "/opt/ros/noetic/lib/python3/dist-packages/actionlib/action_client.py", line 466, in init_goal
    self.send_goal_fn(action_goal)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 882, in publish
    self.impl.publish(data)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 1066, in publish
    serialize_message(b, self.seq, message)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/msg.py", line 152, in serialize_message
    msg.serialize(b)
  File "/home/dennis/Workspaces/binpick_ws/devel/lib/python3/dist-packages/ur_dashboard_msgs/msg/_SetModeActionGoal.py", line 134, in serialize
    buff.write(_get_struct_b2B().pack(_x.goal.target_robot_mode, _x.goal.stop_program, _x.goal.play_program))
AttributeError: 'SetModeActionGoal' object has no attribute 'target_robot_mode'
```

## Steps to Reproduce
I have created a small script that replicates the issue:
```
#!/usr/bin/env python3
import rospy
import actionlib
from ur_dashboard_msgs.msg import SetModeAction, SetModeActionGoal

rospy.init_node('program_action_test', anonymous=True)
action_client = actionlib.SimpleActionClient('/ur_hardware_interface/set_mode',SetModeAction)

if action_client.wait_for_server(rospy.Duration(3)):
    message = SetModeActionGoal()
    message.goal.target_robot_mode = 7
    message.goal.stop_program = False
    message.goal.play_program = True
    action_client.send_goal(message)
```

## Workaround Suggestion
I can manually publish the goal from the terminal as follows, however I cannot use this for what I am building as I need it to trigger from within the program:
```
rostopic pub /ur_hardware_interface/set_mode/goal ur_dashboard_msgs/SetModeActionGoal "goal:
  target_robot_mode: 7
  stop_program: False
  play_program: True"
```

I can also create a Publisher that publishes the SetModeActionGoal to '/ur_hardware_interface/set_mode/goal' and use that. However, this is not reliable as I have to re-publish multiple times before the program actually restarts on the teach pendant.

gavanderhoorn, 2021-03-10:
> ```python
> message = SetModeActionGoal()
> ```

shouldn't this be `SetModeGoal`?

dionysius07, 2021-03-10:
Thank you for the quick response! Evidently it is indeed supposed to be SetModeGoal as opposed to SetModeActionGoal. I used the latter as that was the type that tab-autocompleted in the terminal when I was testing it out via rostopic publication and I assumed that was the one to use. Switching it out to the former works as intended and reliably. Closing this issue.

closed 2021-03-10 by dionysius07