UniversalRobots/Universal_Robots_ROS_Driver #334: URsim docker not working on windows after Docker Desktop update 3.1.0 -> 3.2.0

BlazPotokar, 2021-03-11:
I created docker image with instructions available in folder `.github/dockerursim` and with slight modification as mentioned in issue #324. 
This works fine on Docker Desktop 3.1.0. With an update to the latest version of Docker Desktop, which is 3.2.0, URsim does not work as it should. It doesn't load controller (image bellow). I tested it for UR10 e-series. 

Changes between docker desktop versions can be seen [here](https://docs.docker.com/docker-for-windows/release-notes/), but I don't know what is causing problem. 

I am using docker image to connect it with UR ROS driver in another docker container to simulate robot moves (I need to run it on windows). I use docker image instead of VM because it should be more lightweight. 

[image]

fmauch, 2021-03-11:
Please note that docker use of URSim isn't fully supported. It is used inside the continuous integration only.

However, thanks for reporting this as we might stumble into this in CI at some point.

BlazPotokar, 2021-03-12:
> Please note that docker use of URSim isn't fully supported. It is used inside the continuous integration only.

Can you tell more about which functions are not supported and potential problems that can appear using URsim in docker (like in my pipeline with moveit as master)?

fmauch, 2021-03-13:
I didn't mean "supported" in a way that certain things don't work, but that we don't give support for using URSim inside a docker container, as this would be out of our current maintaining resources.

closed 2021-03-27 by BlazPotokar