UniversalRobots/Universal_Robots_ROS_Driver #336: Beta yaml error

bblumberg, 2021-03-12:
# Summary

Built beta driver following instructions in the melodic docker container from docker hub. Builds fine, but when I do the roslaunch, I get the following error:
roslaunch ur_robot_driver ur3e_bringup.launch robot_ip:=10.0.1.42
... logging to /root/.ros/log/85cd9ccc-8369-11eb-96a9-48b02d2b86b4/roslaunch-bruce-ai-19473.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
RLException: error loading <rosparam> tag: 
	file /ur/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/config/ur3e_controllers.yaml contains invalid YAML:
mapping values are not allowed here
  in "<string>", line 165, column 60:
     ... robot_status_handle publish_rate: 10
                                         ^
XML is <rosparam command="load" file="$(arg controller_config_file)"/>
The traceback for the exception was written to the log file

*Introduction to the issue*

# Versions
 - ROS Driver version: Beta
 - Affected Robot Software Version(s):
 - Affected Robot Hardware Version(s):
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s):

# Impact
*What is the impact of the issue. Try to describe it well enough, in order for us to prioritise the issues.*
Blocker

# Issue details
*This part is important in order to clarify the severity of the issue.*
see above

## Use Case and Setup
*What does the setup look like and what are the objectives?*

## Project status at point of discovered
*When did you first observe the issue?*
 - *Before I started?*
 - *In first couple of tries?*
 - *In normal use?*
 - *etc...*

## Steps to Reproduce
*Make simple example to reproduce the issue. Try to remove dependencies to other hardware and software components, if it is possible.*
Follow instructions, then do ros launch per instructions

## Expected Behavior
*What did you expect and why?*

## Actual Behavior
*What did you observe? If possible please attach relevant information.*

## Workaround Suggestion
*If a workaround has been found, you are welcome to share.*

gavanderhoorn, 2021-03-13:
Problematic line is here:

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/a6b9f5ed85dc32303a0324a3161102b5ef6b9f7b/ur_robot_driver/config/ur3e_controllers.yaml#L165

`publish_rate: 10` should be on its own line.

This was introduced in dd2fe918 as part of #319.

stefanscherzinger, 2021-03-13:
Thanks @gavanderhoorn @bblumberg

fmauch, 2021-03-24:
@stefanscherzinger Is this fixed?

stefanscherzinger, 2021-03-24:
Yes, was fixed in #337.

PR #337 (2021-03-13): Fix yaml file for UR3e robots
This fixes #336

Thanks to @gavanderhoorn @bblumberg
commit 3cb33d00 (2021-03-13): Fix yaml file for UR3e robots (#337)

This fixes #336
closed 2021-03-24 by stefanscherzinger