UniversalRobots/Universal_Robots_ROS_Driver #338: Troubleshoot getCurrentPose values

TuVieja, 2021-03-15:
*Introduction to the issue*

Hi,

I'am working with UR10e-series robot and ROS, and I'm triying to move it with MoveIt. The problem is that when I getCurrentPose();, the values are different from the  TeachPendant. Some of the axis are negative when in reality they're positive or viceversa.

If I get the /tf from rostopic echo /tf, the pose is OK.

# Versions
 - ROS Driver version: ROS Melodic
 - Affected Robot Hardware Version(s):UR10e-series
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s): externalcontrol 1.0.4

fmauch, 2021-03-15:
I guess, we are talking about [this function](http://docs.ros.org/en/melodic/api/moveit_ros_planning_interface/html/classmoveit_1_1planning__interface_1_1MoveGroupInterface.html#a6430022f7709b989857af33b1a128563)?

Watch out for the reference frame of that pose. `base` and `base_link` are rotated which will result in swapped axes when a pose is given in `base_link` in contrast to `base`. Basically, we followed [this guideline](https://gavanderhoorn.github.io/rep/rep-0199.html#coordinate-frames)

Also note, that you will only get the same numbers when `tool0` and `tool0_controller` line up. This will be the case when
 - You are using a calibrated URDF for your specific robot
 - There is no tool configured on the teach panel.

If things do not get any clearer from my short answer, please post the exact commands (code snippets) you are using to generate the statements above.
 - Code snippet with the `getCurrentPose() call` (basically: Which frame is being asked for)
 - The result of this call
 - The tf lookup you made
 - The numbers from the TP

gavanderhoorn, 2021-03-16:
@fmauch wrote:
>     * The tf lookup you made

And to clarify: `rostopic echo /tf` is not the correct command to use. It would be `rosrun tf2_tools echo.py base tool0`.

stefanscherzinger, 2021-08-09:
@TuVieja Feel free to re-open if the recommendations do not fix your issue.

closed 2021-08-09 by stefanscherzinger