UniversalRobots/Universal_Robots_ROS_Driver #339: Move until tool contact

marekcygan, 2021-03-17:
In the e-series UR robots have the nice teach pendant programming command move until tool contact (even with retraction distance defined). Are there any plans to cover this functionality in the driver?

fmauch, 2021-03-17:
Currently, this is not planned.

marekcygan, 2021-03-17:
@fmauch thanks for fast response! Do you recommend some particular way of implementing such a feature on my end? My current plan is to send a piece of UR script with a loop actively checking for `tool_contact` and using `speedl` along the way.

ljden, 2021-05-06:
@marekcygan did you make any progress with this?

marekcygan, 2021-07-02:
@ljden sorry for late reply. In the end we used [ur_rtde](https://sdurobotics.gitlab.io/ur_rtde/) and injected a small ur script that actively checks for contact and triggers `stopl`.

closed 2021-08-09 by stefanscherzinger