UniversalRobots/Universal_Robots_ROS_Driver #347: Multirobot support

Chemaslane, 2021-03-29:
Hi,

I'm trying a setup with two different Universal Robots in the same Ros system, and have been able to launch two instances of the driver at the same time, including the "ur_common.launch" file from different namespaces for each robot.

However, some errors arises when launching the second driver, and the rosnode list becomes something like this

[image]

Is this possible even if I have to make some changes?

fmauch, 2021-03-30:
This might be a duplicate of #81. Did you check that one?

Chemaslane, 2021-03-31:
Ups, yes, that's it. Sorry for the duplicate

closed 2021-03-31 by Chemaslane