UniversalRobots/Universal_Robots_ROS_Driver #348: Which ports to expose when running ur_robot_driver in docker container

robotrapta, 2021-04-01:
# Summary

I'm trying to figure out which ports to expose from a docker container to run `ur_robot_driver`.  Exposing 50001-50002 on both TCP & UDP doesn't work.

# Versions
 - ROS Driver version: Nnoetic 1.15.9, Recent git release of Universal_Robots_ROS_Driver: 98e0d87234cdbd75736c0a30b817cd4ec34bc469 built from source
 - Affected Robot Software Version(s): URSoftware 3.8.0.61336
 - Affected Robot Hardware Version(s): UR5 CB3
 - Robot Serial Number: 2015350556
 - UR+ product(s) installed: none
 - URCaps Software version(s): ros_slave

(Apologies if I didn't get these all right.  Happy to clarify.)

# Impact
I found a reasonable workaround, so it's honestly not that much.  The workaround to run `--network=host` works well enough.  I'm mostly confused about what's going on.  I see with `lsof` that it listens on some other ports, but they seem unstable.

# Issue details

I'm running ur_robot_driver on a docker container - the container has ubuntu 20.04 with ros-noetic, building from source.

```
docker run -it --network=host --privileged pbrain ./run1-urdriver.sh
```
it works.  When I launch the ros-slave URCap I see in the logs

```
[ INFO] [1616712394.450920270]: Robot requested program
[ INFO] [1616712394.451053805]: Sent program to robot
[ INFO] [1616712394.479503497]: Robot ready to receive control commands.
```

end everything works.  But if I run it with restricted network ports:

```
docker run -it \
     -p 50001-50002:50001-50002/udp \
     -p 50001-50002:50001-50002/tcp \
     --privileged pbrain ./run1-urdriver.sh
```

then it starts connecting, but doesn't finish:

```
[ INFO] [1616712209.579851614]: Robot requested program
[ INFO] [1616712209.579988713]: Sent program to robot
```

and I can't actually drive the robot.
 

## Use Case and Setup

I’m running ROS on an nvidia jetson board which only supports Ubuntu 18.04, which means melodic, which means python2, which is a non-starter for us.  Nvidia’s recommendation is to run Ubuntu 20.04 and Noetic in a docker container, which is easy enough, except for issues like this.

## Project status at point of discovered

Code is basically working but we’re trying to deploy it to our target hardware.

## Steps to Reproduce

I could attach a Dockerfile if that’s helpful, but I think the problem should be clear enough as is.

## Expected Behavior

I expected that only ports 50001 and 50002 need to be exposed from the ROS machine (both UDP and TCP).

## Actual Behavior

When I only expose ports 50001 and 50002 we get

```
[ INFO] [1616712209.579851614]: Robot requested program
[ INFO] [1616712209.579988713]: Sent program to robot
```

when starting ros_slave urcap, but not 

```
[ INFO] [1616712394.450920270]: Robot requested program
[ INFO] [1616712394.451053805]: Sent program to robot
[ INFO] [1616712394.479503497]: Robot ready to receive control commands.
```

## Workaround Suggestion

Running the container with `—network=host` works around the problem pretty well, but can lead to other complications.

fmauch, 2021-04-20:
Do you run the robot in a simulation or a real hardware? For the first option, did you open the respective UR interface ports (30001, 30002, 30004, 29999), see [architecture description] there, as well? I guess, you can connect to them, as otherwise you would not even get to the point where you would see those messages.

Looking at the opened ports by the driver:
```
$ ss -alpt | grep ur_robot_driver
LISTEN 0      1                 0.0.0.0:50001          0.0.0.0:*    users:(("ur_robot_driver",pid=180427,fd=18))                                             
LISTEN 0      1                 0.0.0.0:50002          0.0.0.0:*    users:(("ur_robot_driver",pid=180427,fd=16))                                             
LISTEN 0      100               0.0.0.0:50967          0.0.0.0:*    users:(("ur_robot_driver",pid=180427,fd=7))                                              
LISTEN 0      5                 0.0.0.0:40873          0.0.0.0:*    users:(("ur_robot_driver",pid=180427,fd=9))
```

50967 and 40873 are ports being used internally. One is for ROS communication the other one I don't quite know.

You could check the logs on your TP whether that complains about anything.

adrianheron, 2021-05-29:
I was struggling to get a port forwarding scenario working as well and got to the same point of
```
Robot requested program
Sent program to robot
```
but no
```
Robot ready to receive control commands
```

...after banging my head on a wall thinking there were port forwarding issues or socket options or something I stumbled on the real cause by dumb luck  https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/resources/ros_control.urscript#L102

On this line of the script "{{SERVER_IP_REPLACE}}" get replaced with the ip of the server before the script is sent to the controller, but because the server is running inside a docker container its ip is NOT the ip that the controller needs to connect to!  For example if the Host ip is 192.168.1.1 and the docker container ip is 172.27.141.63, 172.27.141.63 is what gets sent in the script and the controller can't connect.  If I modify the script to hardcode 192.168.1.1 then everything starts working.

I'm not sure what the best solution is, maybe expose this as an environment variable or param so that the user can specify it?  Or somehow have it use the ip specified in the External Control urcap instead?

gavanderhoorn, 2021-05-29:
I believe #268 (and following PRs over at the `urcl`) address this.

Kzernobog, 2022-07-05:
Hi guys, a total newbie here when it comes to the UR robots. I was wondering if there was any resolution to this?

I was wondering if it was possible to completely containerise your Robot stack (all the way from drivers to ROS-related stuff, to your experiments) on docker/apptainer? I understand that in a way the above answers already mentioned that it is possible, however, I was wondering what the good software engineering practices around this were. If yes, are their principled approaches to the network resolution problems that often appear when dealing with Robotics and containers? Apologies if this question is too open-ended.

fmauch, 2022-07-11:
To answer the OP:

When running the driver in a docker container, make sure to start the docker container with `-p 50001-50004:50001-50004`.

Then, when starting the driver inside the docker container, use the `reverse_ip:=<IP of docker host>` argument. On the robot, also setup the remote IP to the one of your docker host, in order to use the External Control URCap.

@Kzernobog I hope, this helps.

For the sake of completeness: Expect occasional communication losses when using that kind of setup, as we recommend running the driver on a real-time-enabled PC using a dedicated network device for communicating with the robot.

closed 2022-07-11 by fmauch