UniversalRobots/Universal_Robots_ROS_Driver #349: Compile error building beta-testing branch

robotrapta, 2021-04-03:
# Summary

I can't get the beta-testing branch to build following the README instructions.  My C++ is super rusty (no pun intended) but it seems like the ur_robot_driver package isn't finding the headers from Universal_Robots_Client_Library or something.  The errors are like `urcl::comm::TrajectoryControlMessage has not been declared` even though it's clearly there in the client library code, which is building properly.

# Versions
 - ROS Driver version: tested on both melodic and noetic - identical results
 - Affected Robot Software Version(s): n/a
 - Affected Robot Hardware Version(s): n/a
 - Robot Serial Number: n/a 
 - UR+ product(s) installed: n/a 
 - URCaps Software version(s): n/a

# Impact
Blocked.  Can't use beta-testing branch at all.

## Steps to Reproduce
Following the readme instructions:

```
$ source /opt/ros/melodic/setup.bash
$ mkdir -p catkin_ws/src && cd catkin_ws
$ git clone -b beta-testing-boost https://github.com/UniversalRobots/Universal_Robots_Client_Library.git src/Universal_Robots_Client_Library
$ git clone -b beta-testing https://github.com/UniversalRobots/Universal_Robots_ROS_Driver.git src/Universal_Robots_ROS_Driver
$ git clone -b beta-testing https://github.com/fzi-forschungszentrum-informatik/cartesian_ros_control.git src/cartesian_ros_control
$ git clone -b calibration_devel https://github.com/fmauch/universal_robot.git src/fmauch_universal_robot
$ sudo apt update -qq
$ rosdep update
$ rosdep install --from-paths src --ignore-src -y
$ catkin_make_isolated
```

gives build error messages:

```
/home/astro/beta_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:675:19: error: ‘class urcl::UrDriver’ has no member named ‘writeTrajectoryControlMessage’
       ur_driver_->writeTrajectoryControlMessage(urcl::comm::TrajectoryControlMessage::TRAJECTORY_NOOP);
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/astro/beta_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp: In member function ‘virtual void ur_driver::HardwareInterface::write(const ros::Time&, const ros::Duration&)’:
/home/astro/beta_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:675:19: error: ‘class urcl::UrDriver’ has no member named ‘writeTrajectoryControlMessage’
       ur_driver_->writeTrajectoryControlMessage(urcl::comm::TrajectoryControlMessage::TRAJECTORY_NOOP);
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/astro/beta_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:675:61: error: ‘urcl::comm::TrajectoryControlMessage’ has not been declared
       ur_driver_->writeTrajectoryControlMessage(urcl::comm::TrajectoryControlMessage::TRAJECTORY_NOOP);
                                                             ^~~~~~~~~~~~~~~~~~~~~~~~
/home/astro/beta_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:675:61: error: ‘urcl::comm::TrajectoryControlMessage’ has not been declared
       ur_driver_->writeTrajectoryControlMessage(urcl::comm::TrajectoryControlMessage::TRAJECTORY_NOOP);
                                                             ^~~~~~~~~~~~~~~~~~~~~~~~
/home/astro/beta_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:679:19: error: ‘class urcl::UrDriver’ has no member named ‘writeTrajectoryControlMessage’
       ur_driver_->writeTrajectoryControlMessage(urcl::comm::TrajectoryControlMessage::TRAJECTORY_NOOP);
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/astro/beta_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:679:19: error: ‘class urcl::UrDriver’ has no member named ‘writeTrajectoryControlMessage’
       ur_driver_->writeTrajectoryControlMessage(urcl::comm::TrajectoryControlMessage::TRAJECTORY_NOOP);
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/astro/beta_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:679:61: error: ‘urcl::comm::TrajectoryControlMessage’ has not been declared
       ur_driver_->writeTrajectoryControlMessage(urcl::comm::TrajectoryControlMessage::TRAJECTORY_NOOP);
                                                             ^~~~~~~~~~~~~~~~~~~~~~~~
/home/astro/beta_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:679:61: error: ‘urcl::comm::TrajectoryControlMessage’ has not been declared
       ur_driver_->writeTrajectoryControlMessage(urcl::comm::TrajectoryControlMessage::TRAJECTORY_NOOP);
                                                             ^~~~~~~~~~~~~~~~~~~~~~~~
/home/astro/beta_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:689:91: error: ‘MODE_SPEEDL’ is not a member of urcl::comm::ControlMode’
       ur_driver_->writeJointCommand(cartesian_velocity_command_, urcl::comm::ControlMode::MODE_SPEEDL);
                                                                                           ^~~~~~~~~~~
/home/astro/beta_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:689:91: error: ‘MODE_SPEEDL’ is not a member of urcl::comm::ControlMode’
       ur_driver_->writeJointCommand(cartesian_velocity_command_, urcl::comm::ControlMode::MODE_SPEEDL);
                                                                                           ^~~~~~~~~~~
/home/astro/beta_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:703:87: error: ‘MODE_POSE’ is not a member of ‘urcl::comm::ControlMode’
       ur_driver_->writeJointCommand(cartesian_pose_command_, urcl::comm::ControlMode::MODE_POSE);
                                                                                       ^~~~~~~~~
/home/astro/beta_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:703:87: error: ‘MODE_POSE’ is not a member of ‘urcl::comm::ControlMode’
       ur_driver_->writeJointCommand(cartesian_pose_command_, urcl::comm::ControlMode::MODE_POSE);
[... 41 lines cut ...]
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/astro/beta_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:1214:57: error: ‘urcl::comm::TrajectoryControlMessage’ has not been declared
   ur_driver_->writeTrajectoryControlMessage(urcl::comm::TrajectoryControlMessage::TRAJECTORY_START, point_number);
                                                         ^~~~~~~~~~~~~~~~~~~~~~~~
/home/astro/beta_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:1232:17: error: ‘class urcl::UrDriver’ has no member named ‘writeTrajectoryPoint’
     ur_driver_->writeTrajectoryPoint(p, true, next_time - last_time);
                 ^~~~~~~~~~~~~~~~~~~~
/home/astro/beta_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp: In member function ‘void ur_driver::HardwareInterface::cancelInterpolation()’:
/home/astro/beta_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:1241:15: error: ‘class urcl::UrDriver’ has no member named ‘writeTrajectoryControlMessage’
   ur_driver_->writeTrajectoryControlMessage(urcl::comm::TrajectoryControlMessage::TRAJECTORY_CANCEL);
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/astro/beta_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:1241:57: error: ‘urcl::comm::TrajectoryControlMessage’ has not been declared
   ur_driver_->writeTrajectoryControlMessage(urcl::comm::TrajectoryControlMessage::TRAJECTORY_CANCEL);
                                                         ^~~~~~~~~~~~~~~~~~~~~~~~
CMakeFiles/ur_robot_driver_node.dir/build.make:86: recipe for target 'CMakeFiles/ur_robot_driver_node.dir/src/hardware_interface.cpp.o' failed
```

## Expected Behavior
Successful build.

## Actual Behavior
Build failed.

## Workaround Suggestion
A small change to a build configuration maybe???

fmauch, 2021-04-03:
This should not be the case, but is it possible, that another existing installation of the ur_client_library shadows the one from this workspace?

What is the output of `echo $CMAKE_PREFIX_PATH` (from the shell that your build failed)?

Is the binary debian package `ros-melodic-ur-client-library` installed?

fmauch, 2021-04-03:
It should not be shadowed by the one installed by the apt system... We'll dig deeper.

MikasaIL, 2024-06-06:
Have you guys solved the problem?Please say something~

fmauch, 2024-06-07:
The beta-testing branch has long been deprecated. Uninstalling the binary package when building the library from source seems like  a solid workaround and we haven't found anything else until today.

I'll close this since the branch this is referring to isn't active anymore.

closed 2024-06-07 by fmauch