UniversalRobots/Universal_Robots_ROS_Driver #351: UR3 teaching pendant stopped

mywnajsldkf, 2021-04-20:
# Summary

Hi, I want to connect UR3 hardware with ROS.
But When I start ExternalControl file, teaching pendant screen is stopped. 
How can I solve this problem?

[image]

# Versions
 - ROS Driver version: Ros melodic
 - Affected Robot Software Version(s): 3.7
 - Affected Robot Hardware Version(s): UR3
 - URCaps Software version(s): 1.0.5

fmauch, 2021-05-01:
- Did you setup the URCap with the correct remote IP address as described [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/install_urcap_cb3.md) / Is 192.160.0.100 the IP address of your machine running the ROS driver?
- Did you start the ROS driver before starting the program on the TP?

Anyway, this should not block forever, I'll try to reproduce this.

fmauch, 2024-07-03:
I've never been able to reproduce this and there has never been a follow-up answer, so I'll close this. Feel free to comment / reopen if the issue still persists.

closed 2024-07-03 by fmauch