UniversalRobots/Universal_Robots_ROS_Driver #356: Issue establishing communication between ROS driver and URSim

EspenTeigen, 2021-04-27:
# Summary

Your issue may already be reported!
Please search on the [issue track](../) before creating one.

*Introduction to the issue*

# Versions
 - ROS Driver version:
 - b5e8c56
 - Affected Robot Software Version(s):
 - 3.15.0.106151
 - Affected Robot Hardware Version(s):
 - URSim
 - Robot Serial Number:
 - N/A
 - UR+ product(s) installed:
 - N/A
 - URCaps Software version(s):
 - externalcontrol-1.0.5

# Impact
*What is the impact of the issue. Try to describe it well enough, in order for us to prioritise the issues.*
Research project is halted because of this error

## Use Case and Setup
*What does the setup look like and what are the objectives?*
ROS melodic with UR ROS driver is installed on WSL2, and a URSim of UR10 is running in VmWare

## Steps to Reproduce
*Make simple example to reproduce the issue. Try to remove dependencies to other hardware and software components, if it is possible.*
Install UR driver and URSim with external control URCap, try to run roslaunch ur_bringup ur10_bringup.launch robot_ip:=<UR controllers IP>

## Expected Behavior
*What did you expect and why?*
I expected communication between ROS and URSim.

## Actual Behavior
*What did you observe? If possible please attach relevant information.*
I can see on PolyScope that connection is established, since it goes in to run mode when I start the program on the ROS side, but I do not receive any joint_states. If i try to move the robot I get the message

`Connection closed (command): EOF on recv`

When I start the program I get 

Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://pc7275:52086/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /robot_ip_address: 192.168.78.129
 * /robot_reverse_port: 50001
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /tf2_buffer_server/buffer_size: 120.0
 * /ur_driver/max_payload: 20.0
 * /ur_driver/max_velocity: 10.0
 * /ur_driver/min_payload: 0.0
 * /ur_driver/prefix:

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    tf2_buffer_server (tf2_ros/buffer_server)
    ur_driver (ur_driver/driver.py)

auto-starting new master
process[master]: started with pid [11386]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 306d63d0-a72b-11eb-8e3c-cc483a60737b
process[rosout-1]: started with pid [11397]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [11403]
process[ur_driver-3]: started with pid [11405]
process[tf2_buffer_server-4]: started with pid [11407]
Setting prefix to
[WARN] [1619508953.617002]: No calibration offset for joint "shoulder_pan_joint"
[WARN] [1619508953.619024]: No calibration offset for joint "shoulder_lift_joint"
[WARN] [1619508953.620787]: No calibration offset for joint "elbow_joint"
[WARN] [1619508953.622416]: No calibration offset for joint "wrist_1_joint"
[WARN] [1619508953.624132]: No calibration offset for joint "wrist_2_joint"
[WARN] [1619508953.625802]: No calibration offset for joint "wrist_3_joint"
[INFO] [1619508953.627908]: No calibration offsets loaded from urdf
[INFO] [1619508953.629986]: Max velocity accepted by ur_driver: 10.0 [rad/s]
[INFO] [1619508953.632761]: Bounds for Payload: [0.0, 20.0]
Disconnected.  Reconnecting
[INFO] [1619508953.748285]: Programming the robot
[INFO] [1619508953.752025]: Programming the robot at 192.168.78.129
RobotStateRT has wrong length: 1220
Exception in thread URConnectionRT:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/espente/ur_ws/src/fmauch_universal_robot/ur_driver/src/ur_driver/driver.py", line 372, in __run
    self.__on_packet(packet)
  File "/home/espente/ur_ws/src/fmauch_universal_robot/ur_driver/src/ur_driver/driver.py", line 338, in __on_packet
    for i, q in enumerate(stateRT.q_actual):
AttributeError: 'RobotStateRT' object has no attribute 'q_actual'

Likely a syntax error:
7     URControl10-03-2021, 09:43:57
Exception in thread URConnection:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/espente/ur_ws/src/fmauch_universal_robot/ur_driver/src/ur_driver/driver.py", line 274, in __run
    self.__on_packet(packet)
  File "/home/espente/ur_ws/src/fmauch_universal_robot/ur_driver/src/ur_driver/driver.py", line 186, in __on_packet
    state = RobotState.unpack(buf)
  File "/home/espente/ur_ws/src/fmauch_universal_robot/ur_driver/src/ur_driver/deserialize.py", line 324, in unpack
    raise Exception("Fatal error when unpacking RobotState packet")
Exception: Fatal error when unpacking RobotState packet

Handling a request
[2021-04-27 09:35:54.722136] Out: hello
[INFO] [1619508954.724302]: Robot connected

And when I cancel the program I get

`^C[tf2_buffer_server-4] killing on exit
[ur_driver-3] killing on exit
[robot_state_publisher-2] killing on exit
Traceback (most recent call last):
  File "/home/espente/ur_ws/src/fmauch_universal_robot/ur_driver/src/ur_driver/driver.py", line 1018, in <module>
    if __name__ == '__main__': main()
  File "/home/espente/ur_ws/src/fmauch_universal_robot/ur_driver/src/ur_driver/driver.py", line 1007, in main
    action_server.start()
  File "/home/espente/ur_ws/src/fmauch_universal_robot/ur_driver/src/ur_driver/driver.py", line 692, in start
    self.init_traj_from_robot()
  File "/home/espente/ur_ws/src/fmauch_universal_robot/ur_driver/src/ur_driver/driver.py", line 680, in init_traj_from_robot
    time.sleep(0.1)
KeyboardInterrupt
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done`

This setup worked last week, and this error came after the weekend. I have tried to reinstall the URSim, UR_driver and URCap

## Workaround Suggestion
*If a workaround has been found, you are welcome to share.*

gavanderhoorn, 2021-04-27:
> ```
> ur_driver (ur_driver/driver.py)
> ```

you seem to have started `ur_driver`, which is the old, deprecated driver. Not `ur_robot_driver`.

You didn't show which `.launch` file you started, but I'm going to assume it was not one provided by the `ur_robot_driver` package.

Please make sure you're starting the correct files.

Additionally, I'm not 100% sure you won't run into any issues with network connectivity, seeing as you're running things under WSL and inside a VM. Make sure to check for working connectivity in both directions, on the required ports.

EspenTeigen, 2021-04-27:
Thank you for a swift reply.

Shit, that was embarrassing. It works now.  

What kind of files are used from the fmauch library? Can I just delete the ur_driver directory?

closed 2021-04-27 by EspenTeigen

gavanderhoorn, 2021-04-27:
> Can I just delete the ur_driver directory?

yes.