UniversalRobots/Universal_Robots_ROS_Driver #359: Cmake error at ur_dashboard_msgs

mayday-sauce, 2021-05-01:
**Summary**

Hello,
After installation, the catkin_make command produces an error and configuring is incomplete.  I am new to ROS and I don't know how to proceed or what impact that error has. Thank you in advance for help!

**Versions**
 - ROS Driver version: melodic

Issue details
invoking catkin_make ends with this error: 

CMake Error at /home/klara/catkin_ws/build/Universal_Robots_ROS_Driver/ur_dashboard_msgs/cmake/ur_dashboard_msgs-genmsg.cmake:3 (message):
  Could not find messages which
  '/home/klara/catkin_ws/devel/share/ur_dashboard_msgs/msg/SetModeAction.msg'
  depends on.  Did you forget to specify generate_messages(DEPENDENCIES ...)?

  Cannot locate message [Header] in package [std_msgs] with paths
  [['/home/klara/catkin_ws/devel/share/std_msgs/cmake/../msg']]
Call Stack (most recent call first):
  /opt/ros/melodic/share/genmsg/cmake/genmsg-extras.cmake:307 (include)
  Universal_Robots_ROS_Driver/ur_dashboard_msgs/CMakeLists.txt:79 (generate_messages)

-- Configuring incomplete, errors occurred!
See also "/home/klara/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/klara/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

fmauch, 2021-05-03:
Did you run the `rosdep` commands as explained in the [build section README](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#building)?

The missing message should come from the `std_msgs` package that should be installed. However, I cannot quite imagine you having a ROS installation without `std_msgs`.

What is the output of `echo $ROS_PACKAGE_PATH` in the shell that you tried to build the package in? It should contain `/opt/ros/melodic/share`.

mayday-sauce, 2021-05-03:
I ran the commands, and the output is the following:

klara@klara-HP-250-G6-Notebook-PC:~/catkin_ws$ rosdep install --from-paths src --ignore-src -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ur_robot_driver: Cannot locate rosdep definition for [ur_e_description]

The output of $ROS_PACKAGE_PATH is 
/home/klara/catkin_ws/src:/opt/ros/melodic/share

The output of rospack find std_msgs is
/opt/ros/melodic/share/std_msgs

fmauch, 2021-05-03:
> ur_robot_driver: Cannot locate rosdep definition for [ur_e_description

Looks like you didn't execute that line from the build instructions:

```bash
# clone fork of the description. This is currently necessary, until the changes are merged upstream.
$ git clone -b calibration_devel https://github.com/fmauch/universal_robot.git src/fmauch_universal_robot
```

fmauch, 2021-05-12:
mkay... So can we close this issue?

closed 2021-05-18 by mayday-sauce