UniversalRobots/Universal_Robots_ROS_Driver #360: External control freezes on TP

Robert-Ven, 2021-05-03:
# Summary

Your issue may already be reported!
Please search on the [issue track](../) before creating one.

*Introduction to the issue*
The issue occurs when attempting to connect the UR3 to the ROS pc using external control. After starting the node on the ROS pc, the external control is started, which freezes at the starting program phase.

The problem seems related to [this](https://forum.universal-robots.com/t/external-control-urcap-never-starting-the-program/8095) and [this](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/74#issuecomment-572084667), however, no firewall is enabled.

When attempting `ss -atnp | grep 5000`, the result is the following: 
`LISTEN     0      1            *:50001                    *:*                   users:(("ur_robot_driver",pid=15405,fd=17))
LISTEN     0      1            *:50002                    *:*                   users:(("ur_robot_driver",pid=15405,fd=15))
ESTAB      0      0      192.168.10.6:50001              192.168.10.20:43172               users:(("ur_robot_driver",pid=15405,fd=19))
`
It seems that there is a connection. 

The TP can be unfrozen by terminating the control on the ROS pc side, unplugging the ethernet cable does not unfreeze the system.

fmauch, 2021-05-03:
Possible duplicate of #351

Robert-Ven, 2021-05-03:
To answer the questions you placed there: 

> Did you setup the URCap with the correct remote IP address as described here / Is 192.160.0.100 the IP address of your machine running the ROS driver?

Yes, I believe the code part shows an established connection, I also tried changing this, which resulted in the notification that the ros pc was not present

> Did you start the ROS driver before starting the program on the TP?

Yes, this gave an error that the external ros pc was not present,after starting the ROS node the freezing happened.

closed 2026-02-23 by urfeex