UniversalRobots/Universal_Robots_ROS_Driver #362: URKinematics solver doesn't use set end effector as end effector link/can't change end effector link

UmutUz, 2021-05-04:
Before I made the switch to urkinematics, I made a frame called "right_ee_link." That was supposed to be my end effector with a bit of distance to the actual UR10 tooltip/end effector (to approach objects rather than walk into them). This worked fine with KDL, as in, the interactive marker in RVIZ wouldn't hang on top of the robot model but instead on top of the tf frame ahead of the robot. Also, the robot would not try to "snap" onto the interactive marker, instead the robot would move with the marker the same way as if I never added an extra frame far away from the robot.

Enter urkinematics! The UR10 tries to snap onto the marker and won't move at all when the marker isn't near the robots actual tool tip. Planning works fine, moving works also pretty good but my predefined frame is sadly not recognized as the "actual" end effector. Setting it via MoveIt/Python does print "right_ee_link" as my end effector but the robot will still move the actual tool tip to the given goals.

I am not sure if you guys are going to get what I mean so please ask if I need to clarify. 

ROS Melodic
Ubuntu 18.04

## Expected Behavior
Interactive Marker to be on top of the actual frame and act as if the robots actual end effector is the predefined frame.

## Actual Behavior
Robot tries to snap to the interactive marker rather than being controlled by it from a distance/doesn't recognize the set frame as an end effector. I know that the robots in the pictures are standing in a singularity this isn't the problem here, it's supposed to help show you the behaviour. The first picture shows that the robot only moves when the marker is close to the tool tip, then the robot "snaps" on it. The same movement should be done with the marker being on top of the ### left/right_ee_link frame.

[image]
[image]

gavanderhoorn, 2021-05-05:
With the current implementation of `ur_kinematics`, this is *expected behaviour*.

Similar to IKFast, `ur_kinematics` is generated code/hard-coded and only models the base kinematic chain of a UR. This means: a hard-coded list of joints, hard-coded list of links (ie: transforms) and a hard-coded chain configuration.

You cannot use it to solve for arbitrary other chains (which is sort-of what you're trying to do, as adding another frame to the "normal chain" or asking `ur_kinematics` to solve for an ABB IRB 1200 makes no difference).

See [Tool0 URDF Modifications Break Path Planning](https://answers.ros.org/question/256429) for a similar issue on ROS Answers.

UmutUz, 2021-05-05:
Oh, okay thanks. So it should be possible to change the hardcoded list of links if yes where in the code would that be? I changed the names of the displayed on the pictures so urkinematics would read left/right_tool0 to be on at the "added" frame which "kind of" works. Problem is now changing the base_frame which I am struggling to find inside the code.

Also the reason I want to use urkinematics is, for whatever reason Trac IK isn't as reliable as urkinematics is (probably not well configured). Let's just say Trac IK failing to find solutions isn't rare for my case.

gavanderhoorn, 2021-05-05:
> So it should be possible to change the hardcoded list of links

No, that's not what I wrote.

The solver was written for a specific kinematic structure. You cannot add a transform without changing the FK/IK formulas.

Just changing some names is not going to be sufficient.

> Also the reason I want to use urkinematics is, for whatever reason Trac IK isn't as reliable as urkinematics is (probably not well configured). Let's just say Trac IK failing to find solutions isn't rare for my case.

In my experience Trac IK is *very* reliable, but perhaps I've been lucky.

I'd probably suggest porting the `transform-non-tip-goals-to-tip` from MoveIt to `ur_kinematics`. That would seem to solve your problem while not relying on ad-hoc hacks.

Please do note the outstanding PRs and issues related to IK and `ur_kinematics`: [projects/1](https://github.com/ros-industrial/universal_robot/projects/1).

UmutUz, 2021-05-05:
Thanks a lot for your answers :)

closed 2021-05-05 by UmutUz