UniversalRobots/Universal_Robots_ROS_Driver #363: [Question] Adding end effector to ur5e config

ljden, 2021-05-06:
Just a quick usage question.

If I want to add an end effector to my robot (eg robotiq epick), should I be regenerating the ur5e config with the moveIt setup assistant, or can I extend/utilise the [existing config](https://github.com/fmauch/universal_robot/tree/calibration_devel/ur5_e_moveit_config) to add the end effector. 

Also, is this the best place for questions regarding the UR drivers?

Cheers!

fmauch, 2021-05-06:
The best place to ask this would probably the [ros_industrial/universal_robot](https://github.com/ros-industrial/universal_robot) repository or even better ROS Answers.

To quickly answer your question: You'd best create your own URDF utilizing the UR urdf and create a new moveit configuration from that. It's common procedure to setup a moveit config package for your particular application. The ones in the repo mentioned above are only example configurations for the most simple (non-realistic, as the robot will never be mounted in empty space) scenario.

ljden, 2021-05-06:
Oops, wrong repo. cheers.

I've found a fair few forum posts asking basic usage or conceptual understanding questions relater to moveit/UR driver/URSim that have gone unanswered (on ROS Answers). Is that common or was that just bad luck that those few weren't responded to? 

I need a nice blog post or repo that is a tl;dr of the whole process from controllers to planners to interfaces etc. Something high level to put it all in place. I'm trying to wrap my head around a lot in a short time frame! So many parts to this it is hard to grasp it all...

closed 2021-05-06 by ljden