UniversalRobots/Universal_Robots_ROS_Driver #364: i can't connect PC (Ubuntu 18.04) to UR3(CB3 3.7)

KhuuLuanThanh, 2021-05-08:
i try to connect UR3 robot with PC Ubuntu 18.04(ROS melodic)
i use external control 1.05.urcap and setup successful in CB3 ( v3.7).
IP PC ROS: 192.168.0.128 and IP robot 192.168.0.161 . 2 IP 2 works normally.
But when i run: 
roslaunch ur_robot_driver ur3_bringup.launch robot_ip:=192.168.0.161
an error occurred like image.
i follow tutorial in ROS.wiki and this page but can't do it.
Thanks. wish you a nice day!
[image]
[image]

urrsk, 2021-05-10:
It sounds like a connection issue.
Can you verify that you can  ```ping``` the robot from the ROS side as well as ```ping``` the ROS ip from the robot side?

KhuuLuanThanh, 2021-05-10:
It my problem!
This is image Network i setup to Robot
[image]
( the network is connected)

But on terminal Ubuntu:
 I command: ping 192.168.0.100
It like this
[image]

I use Vmware Ubuntu 18.04 and Wifi. And this is the way i use Ethernet cap connect PC with CB3.
[image]

is it the problem my network or my way connect PC and CB3?
Wish you have a nice day.

fmauch, 2021-05-10:
Clearly, there seems to be an error in your network setup.

First of all, please use [code blocks](https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks) for any shell interaction, this makes it a lot more readable than screenshots.

Second, let's find out your network setup.

`ip addr` should print all configured network interfaces. For your setup to work, I would expect that there would be an interface in the same subnet as the robot machine.

As you are running ubuntu inside a VM (which I would not recommend for performance reasons), you'll have to make sure that your physical network connector is available inside the VM. Then, in your VM make sure to [setup a manual IP address](https://linuxize.com/post/how-to-configure-static-ip-address-on-ubuntu-18-04/#configuring-static-ip-address-on-ubuntu-desktop) that is in the same subnet as your robot controller, e.g. `192.168.0.42`

When this is setup correctly, you should be able to execute `ping 192.168.0.100` correctly.

KhuuLuanThanh, 2021-05-12:
@fmauch @urrsk Thanks for your help so much!
Today i can connect with my robot! it my error about setup network!
Wish you have a nice day!

fmauch, 2021-05-12:
I'll mark this resolved, then. Glad to hear it is working now.

closed 2021-05-12 by fmauch