UniversalRobots/Universal_Robots_ROS_Driver #367: Unstable connection to URSim

ljden, 2021-05-10:
# Summary
Running ROS and URSim within a VM has unstable connection

# Versions
 - ROS Driver version: current master branch
 - Affected Robot Software Version(s): 5.10
 - Affected Robot Hardware Version(s):
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s): 1.0.5

# Impact
Stability of ROS Driver to URSim connection

# Issue details
Running on clean install on Ubuntu 18.04 VM
Installed URSim Polyscope 5.10 within VM
Installed ros Melodic
Installed UR ROS Driver as per README instructions. (Including fmauch/universal_robot).
Change `ROS_WS/src/fmauch_universal_robot/ur5_e_moveit_config/config/controllers.yaml` line 3 `action_ns: follow_joint_trajectory` -> `action_ns: /scaled_pos_joint_traj_controller/follow_joint_trajectory` [(#55 comment)](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/55#issuecomment-562215033)
Load URCap `externalcontrol-1.0.5.urcap` and set vm IP (from ifconfig)
Get IP from Polyscope settings (*NB: 0.0.0.0*) 
Run:
```
roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=0.0.0.0
roslaunch ur5_e_moveit_config ur5_e_moveit_planning_execution.launch limited:=true
roslaunch ur5_e_moveit_config moveit_rviz.launch config:=true
```
✅ Can plan and execute motion in rviz and move robot in URSim
❌ Drops connection frequently and errors occur

## Actual Behavior
```
[ INFO] [1620634745.394921084]: Connection to robot dropped, waiting for new connection.
[ INFO] [1620634745.399383227]: Robot ready to receive control commands.
[ERROR] [1620634745.470755137]: Could not get fresh data package from robot
[ INFO] [1620634745.503329544]: Connection to robot dropped, waiting for new connection.
[ INFO] [1620634745.503941077]: Robot ready to receive control commands.
[ INFO] [1620634747.552573110]: Connection to robot dropped, waiting for new connection.
[ INFO] [1620634747.556853727]: Robot ready to receive control commands.
[ERROR] [1620634748.035244555]: Could not get fresh data package from robot
[ERROR] [1620634748.529363843]: Could not get fresh data package from robot
[ INFO] [1620634749.326112779]: Connection to robot dropped, waiting for new connection.
[ INFO] [1620634749.330559169]: Robot ready to receive control commands.
[ERROR] [1620634754.982709287]: Could not get fresh data package from robot
[ERROR] [1620634766.937607368]: Could not get fresh data package from robot
[ERROR] [1620634767.038744072]: Could not get fresh data package from robot
[ERROR] [1620634767.147528314]: Could not get fresh data package from robot
[ERROR] [1620634767.259383901]: Could not get fresh data package from robot
[ERROR] [1620634774.995134229]: Could not get fresh data package from robot
[ INFO] [1620634780.756342200]: Connection to robot dropped, waiting for new connection.
[ INFO] [1620634780.760647579]: Robot ready to receive control commands.
[ERROR] [1620634781.231625185]: Could not get fresh data package from robot
[ERROR] [1620634784.977755093]: Could not get fresh data package from robot
[ERROR] [1620634790.013236679]: Could not get fresh data package from robot
[ INFO] [1620634799.352680567]: Connection to robot dropped, waiting for new connection.
[ INFO] [1620634799.353380175]: Robot ready to receive control commands.
[ERROR] [1620634799.487222663]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620634799.489532557]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620634799.491756693]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620634799.493887460]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620634799.496010395]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620634799.497971844]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620634799.500848030]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620634799.502943695]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620634799.505094482]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620634799.507053292]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620634799.509161649]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620634799.511256361]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620634799.513428100]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620634799.515443020]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620634799.517551818]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620634799.519720148]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620634799.521695182]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620634799.525986619]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620634799.526260882]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ INFO] [1620634806.594794483]: Connection to robot dropped, waiting for new connection.
[ INFO] [1620634806.598878124]: Robot ready to receive control commands.
[ERROR] [1620634807.077811447]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ INFO] [1620634808.430999174]: Connection to robot dropped, waiting for new connection.
[ INFO] [1620634808.435326156]: Robot ready to receive control commands.
[ERROR] [1620634808.508018274]: Could not get fresh data package from robot
[ERROR] [1620634812.833156033]: Could not get fresh data package from robot
[ERROR] [1620634812.934121646]: Could not get fresh data package from robot
[ERROR] [1620634813.035041003]: Could not get fresh data package from robot
[ERROR] [1620634813.136008311]: Could not get fresh data package from robot
[ERROR] [1620634813.237190379]: Could not get fresh data package from robot
[ INFO] [1620634815.850815647]: Connection to robot dropped, waiting for new connection.
[ INFO] [1620634815.855041500]: Robot ready to receive control commands.
[ERROR] [1620634815.927360042]: Could not get fresh data package from robot
[ INFO] [1620634815.946934115]: Connection to robot dropped, waiting for new connection.
[ INFO] [1620634815.951052999]: Robot ready to receive control commands.
[ERROR] [1620634816.027978400]: Could not get fresh data package from robot
[ INFO] [1620634816.047913015]: Connection to robot dropped, waiting for new connection.
[ INFO] [1620634816.052261417]: Robot ready to receive control commands.
[ERROR] [1620634816.129006049]: Could not get fresh data package from robot
[ INFO] [1620634816.150188482]: Connection to robot dropped, waiting for new connection.
[ INFO] [1620634816.154282267]: Robot ready to receive control commands.
[ERROR] [1620634816.229558604]: Could not get fresh data package from robot
[ INFO] [1620634816.251158619]: Connection to robot dropped, waiting for new connection.
[ INFO] [1620634816.255433743]: Robot ready to receive control commands.
[ERROR] [1620634816.331994276]: Could not get fresh data package from robot
[ERROR] [1620634825.042434267]: Could not get fresh data package from robot
```

If you need any other logging info let me know

ljden, 2021-05-10:
Is there any way to improve stability?

Just to clarify, I'm running a single VM only.

ljden, 2021-05-10:
The VMs resource usage is max 80% of CPU, RAM 40% while planning and executing an rviz moveit trajectory

gavanderhoorn, 2021-05-10:
I don't believe `0.0.0.0` is a valid IP.

ljden, 2021-05-10:
I got it from URSim. I wasn't able to set any other IP address. Possibly a symptom of some deeper issue, ie VM setup... 
[image]

fmauch, 2021-05-10:
To my knowledge 0.0.0.0 is used when opening ports to say that connections from any interface are allowed.

In that particular setup I am not sure how it should be configured correctly. I guess, URSim opens the necessary ports without any IP range restriction, so passing `127.0.0.1` (local address) as an IP address to the ROS driver should also work. But I won't expect any improvement from this alone. But maybe, it is...

What does `ss -apt | grep 50001` yield, when you have the driver running?

gavanderhoorn, 2021-05-10:
There is a difference between passing a hostname/IP address to functions like `bind(..)`, and clients specifying hostnames/IP addresses to `connect(..)`.

The latter is what's happening in the driver.

The correct IP to use when using ursim would be the IP address of the machine running URSim.

URSim does not show valid IP addresses most of the time in the dialog you show @ljden.

ljden, 2021-05-10:
> What does `ss -apt | grep 50001` yield, when you have the driver running?
```
LISTEN 0       1                     0.0.0.0:50001               0.0.0.0:*       users:(("ur_robot_driver",pid=9561,fd=18)) 
```

ljden, 2021-05-10:
> URSim does not show valid IP addresses most of the time in the dialog you show @ljden.
> 
> that dialog works on the real controller, but it doesn't have the required infrastructure on a URSim setup to work correctly. For all intents and purposes, you can't really use that dialog for anything.

Ah makes sense

> However, I would strongly recommend not doing that, as it's uncommon and `robot_ip:=127.0.0.1` conveys intent much better than `robot_ip:=0.0.0.0`.

Gotcha, duly noted. 

Running with `ip:=127.0.0.1` behaves about the same:
```
[ INFO] [1620641104.632693387]: Connection to robot dropped, waiting for new connection.
[ INFO] [1620641104.633017424]: Robot ready to receive control commands.
[ERROR] [1620641104.713817911]: Could not get fresh data package from robot
[ERROR] [1620641104.723289553]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620641111.244316789]: Could not get fresh data package from robot
[ INFO] [1620641134.791123808]: Connection to robot dropped, waiting for new connection.
[ INFO] [1620641134.795329797]: Robot ready to receive control commands.
[ERROR] [1620641134.866726499]: Could not get fresh data package from robot
[ INFO] [1620641134.890971836]: Connection to robot dropped, waiting for new connection.
[ INFO] [1620641134.895038586]: Robot ready to receive control commands.
[ERROR] [1620641134.967211253]: Could not get fresh data package from robot
[ INFO] [1620641134.988157602]: Connection to robot dropped, waiting for new connection.
[ INFO] [1620641134.992438813]: Robot ready to receive control commands.
[ERROR] [1620641135.068015289]: Could not get fresh data package from robot
[ERROR] [1620641135.096704201]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620641135.113717836]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620641135.114274947]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620641135.115959104]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620641135.119868163]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620641135.120742930]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620641135.121572163]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620641135.123162948]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620641135.124659129]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ERROR] [1620641135.126037761]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ INFO] [1620641158.092526734]: Connection to robot dropped, waiting for new connection.
[ INFO] [1620641158.096942287]: Robot ready to receive control commands.
[ERROR] [1620641158.574676332]: Could not get fresh data package from robot
[ERROR] [1620641181.312837280]: Could not get fresh data package from robot
[ERROR] [1620641191.250829102]: Could not get fresh data package from robot
[ERROR] [1620641211.254458802]: Could not get fresh data package from robot
```

fmauch, 2021-05-10:
That would have been my question, whether connection is established over the VM interface (and therefore might get routet through the host) or the loopback interface `127.0.0.1`. However, as the robot seems to connect through the loopback interface, we can probably rule out any unfortunate host interface setup...

ljden, 2021-05-10:
The VM network is currently set to bridged adapter (so I can access real HW), would NAT be any better/worse?

gavanderhoorn, 2021-05-10:
It should not matter.

Personally I would probably check a few things:

 1. see whether `URControl` prints anything suspicious while this sequence of disconnect/connect/disconnect is happening
 2. check the log tab in the Polyscope interface. What does that say?
 3. use wireshark to figure out which side is disconnecting the socket(s)
 4. is there perhaps a firewall active, or some other network-level piece of software which could be interfering (VPN, etc)?

ljden, 2021-05-13:
I was doing a fresh install of the VM, and with a clearer head attempted to solve my networking issues. Looks like it may have been in part the fault of URSim and the net-statistics script that was invalid for Ubuntu 18.04 (see [commit for details](https://github.com/ljden/URSim_Install_Guides/commit/ae6c2206c8cd921020bfe61dc82f12a8c77179f7).

I managed to set a consistent static IP (while maintaining an internet connection) for the VM and URSim picks up this IP (likely from the net-statistics script). Running URSim with ROS UR Driver + moveit config planner + rvis ran smoothly with no comms issues.

I may have done something slightly differently elsewhere but the major change was having the net-statistics script running correctly, so I suspect it was a URSim side issue. 

As such closing off the issue, thanks heaps for you help @gavanderhoorn @fmauch!

ljden, 2021-05-13:
Well... I went back just now to check how much of the resource usage was URSim taking as I suspect it is the main user, and lo and behold the same errors of  `Pipeline producer overflowed! <RTDE Data Pipeline>` and  `Could not get fresh data package from robot` are back... Still likely not the fault of UR ROS Driver so I'll leave this issue closed.

re your question @gavanderhoorn ursim prints an error message about RobotStat messages being queued up:
```
12:39:08.730 ERROR - RobotState messages are queued up: 4 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
12:39:27.673 ERROR - RobotState messages are queued up: 4 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
12:39:38.689 ERROR - RobotState messages are queued up: 3 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
12:39:42.746 ERROR - RobotState messages are queued up: 3 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
12:39:50.325 ERROR - RobotState messages are queued up: 3 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
12:40:00.431 ERROR - RobotState messages are queued up: 3 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
Unable to find anchors defining header, the program will not be split
12:41:05.329 ERROR - RobotState messages are queued up: 3 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
12:41:24.576 ERROR - RobotState messages are queued up: 4 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
12:41:25.637 ERROR - RobotState messages are queued up: 4 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
12:41:28.280 ERROR - RobotState messages are queued up: 4 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
12:41:29.656 ERROR - RobotState messages are queued up: 4 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
12:41:49.488 ERROR - RobotState messages are queued up: 3 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
12:41:49.546 ERROR - RobotState messages are queued up: 3 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
12:42:15.707 ERROR - RobotState messages are queued up: 4 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
12:42:53.494 ERROR - RobotState messages are queued up: 4 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
12:43:42.478 ERROR - RobotState messages are queued up: 3 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
12:43:57.360 ERROR - RobotState messages are queued up: 4 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
12:43:59.426 ERROR - RobotState messages are queued up: 4 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
12:44:15.405 ERROR - RobotState messages are queued up: 3 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
12:45:07.513 ERROR - RobotState messages are queued up: 4 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
12:45:08.228 ERROR - RobotState messages are queued up: 4 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
12:45:23.250 ERROR - RobotState messages are queued up: 4 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}
```

Each time that error message is printed, ROS is also printing it's error messages

ljden, 2021-05-13:
> It should not matter.
> 
> Personally I would probably check a few things:
> 
>     1. see whether `URControl` prints anything suspicious while this sequence of disconnect/connect/disconnect is happening
> 
>     2. check the log tab in the Polyscope interface. What does that say?
> 
>     3. use wireshark to figure out which side is disconnecting the socket(s)
> 
>     4. is there perhaps a firewall active, or some other network-level piece of software which could be interfering (VPN, etc)?
> 
> 
> And:
> 
> > The VMs resource usage is max 80% of CPU, RAM 40% while planning and executing an rviz moveit trajectory
> 
> that seems pretty high to me actually. On a typical system, `ur_robot_driver` uses a max of about 20% CPU for me (most the time spent in deserialising traffic coming from the controller). RViz and MoveIt also aren't _too_ demanding any more, unless you're using software OpenGL (which is not uncommon in VMs).

1. `ERROR - RobotState messages are queued up: 4 {thread: RobotState - PostMan , loggerClass: com.ur.monitor.RobotState$PostMan}`
2. Sample of log:
[image]
3. TBC
4. Firewalls are down and it's running on the local network

gavanderhoorn, 2021-05-13:
I'm not sure why you (and the author of the guide you link to) finds that dialog so important.

You're using a simulated environment, why care about what that dialog tells you?

> I managed to set a consistent static IP (while maintaining an internet connection) for the VM and URSim picks up this IP (likely from the net-statistics script)

Unless something has changed recently (other than requiring an account to be able to download it .. :disappointed:), `URControl` will simply open the required ports on `0.0.0.0`, so using the IP of the host running it will always work.

ljden, 2021-05-13:
To be honest, mostly due to not having a solid grasp of what is and isn't required and the fact that I am having network/stability issues. 
It is possible URSim and ROS only runs stably when the VM can't connect to the network and therefore possibly avoid the latency of going through my router over wifi. With that said it's only an educated guess as Im not 100% sure what is the reason for it working vs not

gavanderhoorn, 2021-05-13:
> It is possible URSim and ROS only runs stably when the VM can't connect to the network and therefore possibly avoid the latency of going through my router over wifi.

Anything is possible (as this is software), but unless you've specifically configured your network to work that way (and your VM is using that configuration as well), I'd assume all traffic destined for local hosts (note I'm not writing `localhost`) should never leave the machine from which it originates.

And "ROS" is not really involved here. The connection is made between `ur_robot_driver_node` and ursim. That's all plain-old TCP networking, no ROS in the mix.

It'd still be interesting to see a wireshark capture, to see which side (tries to) tear(s) down the connection, causing these reconnection log messages.

It'd also be interesting to know whether other versions of ursim exhibit the same symptom.

Does a regular simulated CB3 (so not e-Series) also disconnect and reconnect all the time for instance?

ljden, 2021-05-13:
It's been a while since I used wireshark, can I get a tldr of what commands to run/what to look for? 

Haven't tested any other versions of URSim or robot series.
I'm setting up for a uni course so don't have time to investigate others at the moment (unless they show more promise, albeit need to be e-series), but can help investigate these specific issues to properly fix the stability issues.

gavanderhoorn, 2021-05-13:
> can I get a tldr of what commands to run/what to look for?

All in your VM:

```
sudo apt update
sudo apt install wireshark-gtk
# allow non-root users to make captures in the dialog that follows during install
```

now start wireshark.

start a capture on the `any` device.

run what you normally run (start the driver, etc).

stop the capture and save it.

Compress and attach it to the issue here.

ljden, 2021-05-13:
Cheers. I have a longer capture if you need more but I assume this should be heaps.
[ursim_ros_driver_dump.zip](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/6472184/ursim_ros_driver_dump.zip)

gavanderhoorn, 2021-05-13:
The capture shows a lot of spurious retransmissions in bursts of a couple hundred packets each time.

This could indicate thread starvation of some sort in the driver.

If you can, try configuring ursim for a regular CB3 controller instead of the e-Series one, as that will run at 125 Hz instead of 500 Hz.

It could be the VM is just too resource constraint for the driver to be able to properly process all the incoming traffic at 500 Hz.

ljden, 2021-05-15:
I think it is network related as the connection runs stably if the VM is not connected to the network, with no other changes.

Can't spend the time testing a CB3 controller at this point. Installing a new version of URSim would be an hour or so of work as it must be installed prior to ROS (dependencies... ) so I would have to create a new VM instance and get it all set up with the Sim then ROS. Plus we have e-series hardware so we need the VM to be the same.

Appreciate the help though.

closed 2023-05-05 by github-actions[bot]