UniversalRobots/Universal_Robots_ROS_Driver #368: tf_prefix partially ignored

fennel-labs, 2021-05-10:
# Summary
The tf_prefix argument does not have any effect on the generated robot model. It is just used to adjust the frames in the hardware_interface, but never passed down to the launch-file, which generates the robot_description parameter.

# Versions
 - ROS Driver version:
 - Affected Robot Software Version(s): Any
 - Affected Robot Hardware Version(s): Any e-Series, but probably also non-e-series

# Impact
Multiple robots cannot be used. And what is maybe more important: If "base_link" is already taken by a moving platform or any other robot, the UR driver will redefine it.

# Issue details
The problem originates from ur_robot_driver/launch/ur_common.launch, as the tf_prefix argument is not forwarded to the robot description generator:
```
  <include file="$(arg robot_description_file)">
    <arg name="limited" value="$(arg limited)"/>
    <arg name="kinematics_config" value="$(arg kinematics_config)"/>
  </include>
```
Furthermore, fmauch_universal_robot/ur_e_description/urdf/ur10e_robot.urdf.xacro does not take a prefix as an argument.

## Use Case and Setup
In general, the problem occurs, when a complex robotic setup is used, where multiple URDF-models are used and spread among different robot model variables. (E.g. if we have two independent UR robot arms in the same ROS environment).

## Project status at point of discovered
Was always present.

## Steps to Reproduce
Launch `roslaunch ur_robot_driver ur16e_bringup.launch robot_ip:=111.222.111.222 tf_prefix:=prefix` and observe the tf-tree. It will not show the tf_prefix.

## Expected Behavior
tf_prefix is reflected in tf_tree.

## Actual Behavior
tf_prefix is mostly ignored in tf_tree.

## Workaround Suggestion
A workaround is to duplicate the robot description, i.e., `fmauch_universal_robot/ur_e_description/urdf/ur10e_robot.urdf.xacro`, and the launch-file that calls xacro, i.e. `fmauch_universal_robot/ur_e_description/launch/ur10e_upload.launch` and to add the tf_prefix as argument in both. The duplicates can then be invoked using the robot_description_file-argument without touching the driver code.

A more sustainable alternative is to add the changes directly in the repo-files. I would offer to do this myself, but I'm not sure about the process since the driver is split into different repos.

fmauch, 2021-05-12:
Thanks for reporting this. I'll try to address this asap.

fmauch, 2021-05-15:
I found the time to look at this in more detail. In my opinion, the problems you describe should be handled differently:

> Multiple robots cannot be used.

When doing a multi-robot setup, you usually define a new environment description containing both robots, anyway. This is loaded separately from the driver, see below.

> And what is maybe more important: If "base_link" is already taken by a moving platform or any other robot, the UR driver will redefine it.

This indicates that the there is already a robot description on the parameter server. Loading the robot description using the above mentioned launchfile will result in overriding the existing description, which is probably not the desired behavior.

---

Both scenarios mentioned imply the same solution. As the robot description of the full system consists of more than just the one robotic manipulator that shell get started using the `urXX_bringup.launch` launchfile.

In such situations it is better to load the description and start a `robot_state_publisher` separately and then start the robot by creating a launchfile parametrizing the `ur_control.launch` file for your specific application.

Nevertheless I agree that it might make sense to forward the `tf_prefix` parameter to the default description's launchfile (though it wouldn't make sense to add a `tf_prefix` in most cases when it would make sense to load the description that way for the reasons mentioned above). However, I'd not like to further change my fork of the description package, as this will further postpone that from getting merged upstream.

fennel-labs, 2021-05-17:
I general, I agree that scenarios, where multiple are used, should come with their own URDF-file with appropriate prefixes and names. However, there are some scenarios where this is more difficult. An example of this are robots that come with their own robot description package such as the Husky and that shall operate independently within the same ROS master. If two instances of that robot are to be spawned in the same environment, each one will bring its one robot_description-parameter due to the design choice made by clearpath.

fmauch, 2021-05-17:
> I general, I agree that scenarios, where multiple are used, should come with their own URDF-file with appropriate prefixes and names. However, there are some scenarios where this is more difficult. [...]

Yes, clearpath is an example of handling things a bit different. But, as well, they provide mechanisms to extend the platform's description using the [`urdf_extras`](https://github.com/husky/husky/blob/34bd30535f11aad0092ad1217aa691621a29cc75/husky_description/urdf/husky.urdf.xacro#L41) argument inside their description.

The only scenario that I can think of, is to use namespaced robot descriptions, where each robot gets a description parameter in its own namespace. In those cases it would make sense to add a tf_prefix to those descriptions, so the TFs can be broadcasted to a single tf topic.

fmauch, 2023-04-06:
I don't feel like this should be closed already.

closed 2023-08-05 by github-actions[bot]