UniversalRobots/Universal_Robots_ROS_Driver #371: how to use the joint_group_vel_controller

sunnymints, 2021-05-12:
I want to use the joint_group_vel_controller on real robot and gazebo.
This my yall for gazebo:
joint_group_vel_controller:
   type: velocity_controllers/JointGroupVelocityController
   joints:  
     - shoulder_pan_joint
     - shoulder_lift_joint
     - elbow_joint
     - wrist_1_joint
     - wrist_2_joint
     - wrist_3_joint
However I can not use it. I can not found the controller's topic to send command and connect the moveit.

gavanderhoorn, 2021-05-12:
MoveIt is incapable of using a `JointGroupVelocityController`, or are you referring to MoveIt Servo?

If yes:

>  can not found the controller's topic to send command [..]

have you made sure to start the controller?

sunnymints, 2021-05-12:
> MoveIt is incapable of using a `JointGroupVelocityController`, or are you referring to MoveIt Servo?
> 
> If yes:
> 
> > can not found the controller's topic to send command [..]
> 
> have you made sure to start the controller?

thanks for your answer, i  try it again. You are true. I miss the topic: /joint_group_vel_controller/command because i looking in topic monitor of rqt at  beginning. But,  I face a new problem when i start the  `JointGroupVelocityController` the robot urdf mode 
will fall down like losing gravity. I don not know how to solve the problem.

sunnymints, 2021-05-13:
> MoveIt is incapable of using a `JointGroupVelocityController`, or are you referring to MoveIt Servo?
> 
> If yes:
> 
> > can not found the controller's topic to send command [..]
> 
> have you made sure to start the controller?

Hi, I need your help. I cannot solve  this problem when the JointGroupVelocityController is starting, the robot urdf mode
will fall down like losing gravity[*-*]. Thank you very much!

gavanderhoorn, 2021-05-13:
Seeing as you mention Gazebo, this is no longer an issue with the packages in this repository.

When using Gazebo, you do not use `ur_robot_driver` any more, but regular `gazebo_ros_control`.

You should probably close this issue in that case.

As to your problem: I'm not aware of Gazebo performing gravity compensation for you, so unless you are commanding the virtual robot to compensate for gravity, it's going to collapse I believe.

sunnymints, 2021-05-17:
> Seeing as you mention Gazebo, this is no longer an issue with the packages in this repository.
>
> When using Gazebo, you do not use `ur_robot_driver` any more, but regular `gazebo_ros_control`.
>
> You [...]
Thank you very much!
I try to use JointGroupVelocityController on real robot, the robot not fall down. But, the real robot can not get set speed， when we set the end velocity is 0.1, the real robot will run on 0.15 and when we set 0.5, real robot maybe is 0.09 (we set the demonstrator speed is 100%).

About the gazebo, maybe the JointGroupVelocityController can not offer gazebo Plug-in support，this problem will be solved, in the future.

Thank you, again.

sunnymints, 2021-05-17:
> Seeing as you mention Gazebo, this is no longer an issue with the packages in this repository.
>
> When using Gazebo, you do not use `ur_robot_driver` any more, but regular `gazebo_ros_control`.
>
> You [...]

Thank you very much!
I try to use JointGroupVelocityController on real robot, the robot not fall down. But, the real robot can not get set speed， when we set the end velocity is 0.1, the real robot will run on 0.15 and when we set 0.5, real robot maybe is 0.09 (we set the demonstrator speed is 100%).

About the gazebo, maybe the JointGroupVelocityController can not offer gazebo Plug-in support，this problem will be solved in the future.

Thank you, again.

fmauch, 2021-05-17:
For reference, this is a cross-post to https://forum.universal-robots.com/t/ur5-constant-motion/14305/4

sunnymints, 2021-05-18:
https://github.com/ros-industrial/universal_robot/issues/462

closed 2021-05-18 by sunnymints