UniversalRobots/Universal_Robots_ROS_Driver #373: Guide for Cartesian Control

mathujeyanth, 2021-05-12:
# Summary
Hi, I'm using the beta-testing branch together with URsim to test the cartesian control. I am unable to find any topics to perform cartesian control. I suspect it is something that has to be configured.
Could you provide a guide (or refer to somewhere) on how to use the cartesian control using the driver?

## Steps to Reproduce
Follow the build from source from beta-testing branch

## Expected Behavior
Able to control robot with cartesian control out-of-the-box

## Actual Behavior
No cartesian control topics.

stefanscherzinger, 2021-05-12:
Hi @mathujeyanth 

Tutorials are currently in the making. Try the _pose_based_cartesian_traj_controller_. It should be loaded (but stopped) by default. After activating this ROS-controller via the controller manager, you can send the robot Cartesian trajectories by publishing to the respective _follow_cartesian_trajectory_ topic.
```bash
$ rostopic list | grep follow_cartesian
```
[Here](https://github.com/fzi-forschungszentrum-informatik/cartesian_ros_control/blob/beta-testing/pass_through_controllers/examples/script/cartesian_trajectory_action_client.py#L73)'s a Python script that uses this interface. But be careful with driving to random poses on real hardware!

gavanderhoorn, 2021-05-12:
> you can send the robot Cartesian trajectories by publishing

@stefanscherzinger: the script you link seems to create an action client, it doesn't publish trajectories to any topic directly.

I know this is pedantic, but there are `ros_control` controllers which actually accept `JointTrajectory` messages directly on a `command` topic, so it'd be good to be specific about this.

stefanscherzinger, 2021-05-12:
That's true. My bad. The _pose_based_cartesian_traj_controller_, _the joint_based_cartesian_traj_controller_ and the _forward_cartesian_traj_controller_ all provide an action interface for trajectory execution.

mathujeyanth, 2021-05-12:
Hi to both of you, @stefanscherzinger I am able to see the topics now, but not able to move the robot using the "cartesian_trajectory_action_client.py". My procedure currently is to start the URsim and ros driver, afterwards I execute the following to start the pose controller:
```bash
$ rosrun controller_manager controller_manager spawn pose_based_cartesian_traj_controller
```
After it has been starting I execute, though I get stuck at `self.client.wait_for_server()` at line 29.
```bash
$ rosrun pass_through_controllers cartesian_trajectory_action_client.py
```
Anything I'm missing?

stefanscherzinger, 2021-05-12:
> Anything I'm missing?

@mathujeyanth Sorry if my comment was misleading. The script was rather meant as a starting point for your own script. If you want to use that, you should change [this line](https://github.com/fzi-forschungszentrum-informatik/cartesian_ros_control/blob/beta-testing/pass_through_controllers/examples/script/cartesian_trajectory_action_client.py#L27) with what you have on your setup. Something like
```python
         '/pose_based_cartesian_traj_controller/follow_cartesian_trajectory',
```

mathujeyanth, 2021-05-13:
@stefanscherzinger, my bad, seems I did not look properly at the script. Though, I'm still not able to make it work after changeing the line you suggested. I receive the following error:
```bash
$ rosrun pass_through_controllers cartesian_trajectory_action_client.py
Result: error_code: 0
error_string: ''
```
After investigating it seemed like PyKDL from [`random_point()`](https://github.com/fzi-forschungszentrum-informatik/cartesian_ros_control/blob/b35e85e31bfb94dfd027e2da1c33fa9c05893335/pass_through_controllers/examples/script/cartesian_trajectory_action_client.py#L48) was the culprit, since it kept spitting empty transformation matrices. I therefore gave `p1` and `p2` valid pose values (using the /`tf` topic), but still to no avail.

stefanscherzinger, 2021-05-15:
> Result: error_code: 0
error_string: ''

That looks like success. If the robot is still not moving, can you confirm that the correct controller is really running by checking with a service call to the controller manager (`controller_manager/list_controllers`)?

mathujeyanth, 2021-05-18:
Hi, I have (kinda) found a solution to the problem. What worked for me was stopping the scaled joint controller as soon as starting the ros driver using:
```bash
$rosrun controller_manager controller_manager stop scaled_pos_joint_traj_controller
```
followed by 
```bash
$rosrun controller_manager controller_manager spawn pose_based_cartesian_traj_controller
```
Now I'm able to execute the "cartesian_trajectory_action_client.py" script. This also solved my [additional issue](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/373#issuecomment-839841616) of not being able to start the twist controller.

---

Solving this problem lead me to a another problem, where I sometimes simply cant send new cartesian poses. Sometimes it works after stopping and playing from the dashboard two times. Sometimes it works after starting the `pose_based_cartesian_traj_controller` again. Though, as far as I have tested it always work the first time when starting the driver.

---

Lastly as you requested here is the result of the service call to the control manager:
```bash
$rosservice call controller_manager/list_controllers
controller:
  -
    name: "pos_joint_traj_controller"
    state: "initialized"
    type: "position_controllers/JointTrajectoryController"
    claimed_resources:
      -
        hardware_interface: "hardware_interface::PositionJointInterface"
        resources:
          - elbow_joint
          - shoulder_lift_joint
          - shoulder_pan_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint
  -
    name: "joint_group_vel_controller"
    state: "initialized"
    type: "velocity_controllers/JointGroupVelocityController"
    claimed_resources:
      -
        hardware_interface: "hardware_interface::VelocityJointInterface"
        resources:
          - elbow_joint
          - shoulder_lift_joint
          - shoulder_pan_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint
  -
    name: "joint_state_controller"
    state: "running"
    type: "joint_state_controller/JointStateController"
    claimed_resources:
      -
        hardware_interface: "hardware_interface::JointStateInterface"
        resources: []
  -
    name: "scaled_pos_joint_traj_controller"
    state: "running"
    type: "position_controllers/ScaledJointTrajectoryController"
    claimed_resources:
      -
        hardware_interface: "ur_controllers::ScaledPositionJointInterface"
        resources:
          - elbow_joint
          - shoulder_lift_joint
          - shoulder_pan_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint
  -
    name: "speed_scaling_state_controller"
    state: "running"
    type: "ur_controllers/SpeedScalingStateController"
    claimed_resources:
      -
        hardware_interface: "ur_controllers::SpeedScalingInterface"
        resources: []
  -
    name: "force_torque_sensor_controller"
    state: "running"
    type: "force_torque_sensor_controller/ForceTorqueSensorController"
    claimed_resources:
      -
        hardware_interface: "hardware_interface::ForceTorqueSensorInterface"
        resources: []
  -
    name: "pose_based_cartesian_traj_controller"
    state: "running"
    type: "pose_controllers/CartesianTrajectoryController"
    claimed_resources:
      -
        hardware_interface: "cartesian_ros_control::PoseCommandInterface"
        resources:
          - tool0_controller
```

stefanscherzinger, 2021-05-18:
> Now I'm able to execute the "cartesian_trajectory_action_client.py" script

Alright, that's a start. Regarding your way of switching controllers, I personally like the controller manager's _switch_controllers_ service call. That offers a clear interface to start and stop ROS-controllers and could be an alternative to the console-based `rosrun controller_manager ..` approach.

What irritates me a little is that both

> name: "scaled_pos_joint_traj_controller"
    state: "running"

and 

> name: "pose_based_cartesian_traj_controller"
   state: "running"

are running in parallel. Although that's in fact possible with the current implementation of resource handling with that variant of the Cartesian trajectory controller, it could be a possible pitfall. Could you stop the _scaled_pos_joint_traj_controller_ and keep only the _pose_based_cartesian_traj_controller_ running to check whether that solves the problem of

> sometimes simply cant send new cartesian poses

The [current driver implementation](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/beta-testing/ur_robot_driver/src/hardware_interface.cpp#L665) gives preference to the _scaled_pos_joint_traj_controller_ (position_controllers type). So, I guess you must be more than lucky to get through with Cartesian goals while both controllers are running.

That's in fact something to address in our implementation of the resource handling of the cartesian_trajectory_controller.

mathujeyanth, 2021-05-20:
> I personally like the controller manager's _switch_controllers_ service call. That offers a clear interface to start and stop ROS-controllers and could be an alternative to the console-based `rosrun controller_manager ..` approach.

Using the approach seems to solve all of the issue, where I explicitly execute the following:
```bash
$rosservice call /controller_manager/switch_controller "start_controllers:['pose_based_cartesian_traj_controller'] stop_controllers: ['scaled_pos_joint_traj_controller'] strictness: 2 start_asap: false timeout: 0.0"
```
I am now reliably able to set the cartesian trajectory, where I'm not experiencing any of the aforementioned issues.
Thank you, I'll be closing this issue.

closed 2021-05-20 by mathujeyanth