UniversalRobots/Universal_Robots_ROS_Driver #379: Error with 5G conection

pablotrelis, 2021-05-19:
# Summary

 have script that controls UR5e using haptic device, it works perfectly connected directly using ethernet, but it doesn't work when i use 2 5G routers (arround 70ms ping). When i launch ur5e_bringup it gives me that error:

```
[ERROR] [1621327595.576990704]: Could not get fresh data package from robot
[ERROR] [1621327596.989480803]: Can't accept new commands. Controller is not running.
```
Im using scaled_pos_traj_controller and it stopped shortly after launching the bringup.

# Versions
 - Affected Robot Software Version(s): URsim 5.9
 - Affected Robot Hardware Version(s): UR5e
 - URCaps Software version(s):  1.0.5

gavanderhoorn, 2021-05-19:
Isn't this a duplicate/cross-post of [Error with real UR5e connected via 5G](https://answers.ros.org/question/378503) on ROS Answers?

pablotrelis, 2021-05-19:
Sorry, I thought that here more people could solve my problem. I proceed to close the issue

closed 2021-05-19 by pablotrelis