UniversalRobots/Universal_Robots_ROS_Driver #383: lowlatency ubuntu kernel

gecastro, 2021-05-21:
# Summary

Regarding the use of a real-time kernel, I noticed that Ubuntu provided a lowlatency version. I understand that this kernel has the PREEMPT_RT patch, although it does not have all the real-time features.

Would anyone recommend the use of this kernel?

For now, I'm looking more at something like a "quick fix" before moving into compiling the kernel for a more permanent solution.

For example, in ubuntu 18.04 I can see the package linux-image-5.4.0-73-lowlatency

Thanks

fmauch, 2021-05-31:
I am unsure what to do with this issue. @gecastro has your question been answered sufficiently? Is there anything missing?

gecastro, 2021-06-01:
Thanks, that answered the question.

closed 2021-06-01 by gecastro