UniversalRobots/Universal_Robots_ROS_Driver #384: ros dependency problem

seunghwa9118, 2021-05-22:
# Summary
I enter on terminer : rosdep install --from-paths src --ignore-src -y
and I got error
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ur_calibration: Cannot locate rosdep definition for [ur_client_library]
ur_robot_driver: Cannot locate rosdep definition for [ur_msgs]

How can i fix it?

# Versions
 - ROS Driver version: kinetic+ 16.04 ubuntu

[image]

Ashok93, 2021-05-23:
I had the same issue. Using `RUN rosdep update --include-eol-distros` fixed it.

lianghongzhuo, 2021-05-23:
The new universal_robot repo: https://github.com/fmauch/universal_robot/tree/calibration_devel
deleted ur_msgs package.

So you can install it by yourself `git clone https://github.com/ros-industrial/ur_msgs.git`

lianghongzhuo, 2021-05-23:
This is needed if you are using noetic, as `ur_msgs` didn't release at noetic version

seunghwa9118, 2021-05-24:
> The new universal_robot repo: https://github.com/fmauch/universal_robot/tree/calibration_devel
> deleted ur_msgs package.
> 
> So you can install it by yourself `git clone https://github.com/ros-industrial/ur_msgs.git`

thank you i fixed it. i just write on the terminal git clone https://github.com/ros-industrial/ur_msgs.git

seunghwa9118, 2021-05-25:
> rosdep update --include-eol-distros

thank you i fixed it

fmauch, 2021-05-31:
I think, this is fixed with #387. Also `ur_msgs` is on the way for ROS noetic. Therefore, I'm closing this.

closed 2021-05-31 by fmauch