UniversalRobots/Universal_Robots_ROS_Driver #385: error: ‘ur_msgs::SetPayload::Request {aka struct ur_msgs::SetPayloadRequest_<std::allocator<void> >}’ has no member named ‘center_of_gravity’

seunghwa9118, 2021-05-22:
# Summary

I doing catkin_make and got error in this picture. how can i fix it?

*Introduction to the issue*

# Versions
 - ROS Driver version: 16.04 ubuntu + kinetic
 
[image]

lianghongzhuo, 2021-05-23:
The new universal_robot repo: https://github.com/fmauch/universal_robot/tree/calibration_devel
deleted ur_msgs package.
Maybe we should add something in the readme?
Like:
`git clone https://github.com/ros-industrial/ur_msgs.git`

fmauch, 2021-05-25:
the `rosdep install` command should make sure to install all necessary dependencies including `ur_msgs`

stefanscherzinger, 2021-08-09:
This has been addressed by #387 and is now in the documentation.

closed 2021-08-09 by stefanscherzinger