UniversalRobots/Universal_Robots_ROS_Driver #386: roslaunch ur_robot_driver ur3_bringup.launch error

seunghwa9118, 2021-05-25:
# Summary
I write on terminal ur_robot_driver ur3_bringup.launch robot_ip:=192.168.100.1 but i got an error
How can i fix it?
# Versions
 - ROS Driver version: 16.04ubuntu + kinetic
[image]

fmauch, 2021-05-25:
Please use [code blocks](https://docs.github.com/en/github/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks) instead of screenshots to attach terminal output.

Did you make sure that your full workspace is up-to date? #97 was merged, an updated `ur_description` (`git pull` inside the `universal_robot` package) is needed.

stefanscherzinger, 2021-08-09:
@seunghwa9118 Is this still an issue after applying @fmauch 's suggestions?

stefanscherzinger, 2021-09-01:
Feel free to re-open if this issue persists.

closed 2021-09-01 by stefanscherzinger