UniversalRobots/Universal_Robots_ROS_Driver #390: robot slowly moving without commands when pose_based_cartesian_traj_controller is running

A13X15, 2021-05-27:
# Summary
When using the controller pose_based_cartesian_traj_controller from the beta-testing branch, the robot is not holding its position when there are no commands, but moving slowly over time. May be a bug in the beta-software.

*Introduction to the issue*

# Versions
 - ROS Driver version: beta-testing, version from May 4th 2021
 - Affected Robot Software Version(s): 3.12
 - Affected Robot Hardware Version(s): UR10
 - Robot Serial Number:
 - UR+ product(s) installed: -
 - URCaps Software version(s): 1.0.4

# Impact
Robot may crash over time, if left unmonitored.

# Issue details

## Use Case and Setup
Tested the beta-branch with a real robot as cartesian control is a neat feature to have. Encoutered what I think is a bug with the 
pose_based_cartesian_traj_controller.

## Project status at point of discovered
 - followed the instructions to install the beta-testing branch 
 - read issue #373 to know how to use the new features
 - modified cartesian_trajectory_action_client.py to test the cartesian movement without random poses...
 - switched controller from scaled_pos_joint_traj_controller to pose_based_cartesian_traj_controller
 - test runs fine, cartesian movement is possible
 - however, robot moves after the script is finished (very slowly, but after a few minutes it becomes obvious)
 - only happen, when pose_based_cartesian_traj_controller is running; not happening without controller or with scaled_pos_joint_traj_controller running

## Steps to Reproduce
- use real hardware for testing (not sure how the simulation behaves, I used real hardware)
- start the pose_based_cartesian_traj_controller
- observe, what the robot is doing over 5-10 minutes

## Expected Behavior
- robot is not moving when there are no commands with pose_based_cartesian_traj_controller running

## Actual Behavior
- robot is slowly moving when there are no commands with pose_based_cartesian_traj_controller running

## Workaround Suggestion
- Idk, if it is a bug, fix it

A13X15, 2021-05-27:
Addition: When watching the teach pendant, I can also see the coordinates and angles changing slowly while the pose_based_cartesian_traj_controller is running.

It would be nice if someone can also test this to confirm, to make sure it is not an issue with my setup, but with the beta-software

fmauch, 2021-05-28:
@stefanscherzinger assigning this to you is OK?

stefanscherzinger, 2021-05-28:
@fmauch That's alright. I'll see how to get hands on a real robot in the next days.

KIT-IFL, 2021-06-29:
Hey @stefanscherzinger , A13X15 is a HiWi at out institute (KIT-IFL) and discovered the bug.

Since we rely heavily on UR's for our research work, we are really keen on testing out the cartesian control.

If you need a real robot you can always drop by, we are very close by the FZI :)
I think A13X15 can even demonstrate the issue for you. Just send me an email:
maximilian.ries@kit.edu

stefanscherzinger, 2021-06-30:
Hi Maximilian (@KIT-IFL), hi @A13X15 

Thanks for the offer! The problem, however, is rather time, not so much available hardware. Without having a proper solution right now, would it maybe be possible for you to check if the problem persists with the new [staging branch](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/tree/staging)? That is the latest state and will be merged into main shortly.

stefanscherzinger, 2021-07-01:
> sure we can try it out

Thanks @A13X15 , that would be perfect!

Until all dependencies are merged upstream, I guess an all-source build is currently the way to go. You need
- [The driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/tree/staging) (staging)
- [ur_msgs](https://github.com/ros-industrial/ur_msgs/tree/melodic-devel) (melodic-devel)
- [ur_client_library](https://github.com/UniversalRobots/Universal_Robots_Client_Library/tree/boost) (boost)

and these four repositories:
- [cartesian_control_msgs](https://github.com/UniversalRobots/Universal_Robots_ROS_cartesian_control_msgs) (main)
- [ros_controllers_cartesian](https://github.com/UniversalRobots/Universal_Robots_ROS_controllers_cartesian) (main)
- [pass_through_controllers](https://github.com/UniversalRobots/Universal_Robots_ROS_passthrough_controllers) (main)
- [scaled_controllers](https://github.com/UniversalRobots/Universal_Robots_ROS_scaled_controllers) (main)

Maybe it's best to setup a new workspace and keep your current configuration untouched.

> I'd assume the pose_based_cartesian_traj_controller is also included in the staging-branch,

Yes, that's included. E.g. for the UR10 [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/staging/ur_robot_driver/config/ur10_controllers.yaml#L147).

A13X15, 2021-07-06:
Hello @stefanscherzinger ,

I just tested the new staging-branch. As you suggested, I set up a new workspace with the repos and dependencies you described, had no problems with building it. For starting the driver, I used an existing launch-file, starting the driver with the correct IP of the robot, YAML-calibration, etc.
When starting said launch-file, I ran into an error because our launch-file uses `load_ur10.launch` from the package `ur_description`, which was not in the workspace. Pulling [universal_robot by fmauch](https://github.com/fmauch/universal_robot/tree/calibration_devel) (calibration_devel) fixed the issue.

The issue with the robot slowly moving when `pose_based_cartesian_traj_controller` is running remains unchanged on the staging-branch. Behavior is exactly the same as described in my first report of the issue using the beta-testing-branch, as soon as `pose_based_cartesian_traj_controller` is running, the robot starts to move at a very slow velocity (like ~1mm/5s, really slow, but clearly visible over long time or when watching the pose/angles on the teach pendant change).

As I'm still the only one reporting on the issue, maybe others can try it out too, in order to rule out the possibility of this being an issue with the setup I'm using.

fmauch, 2021-07-06:
Thanks for testing this out again. We'll have a deeper look into that.

urmahp, 2021-07-08:
I have tested this issue against a simulated robot and against a physical ur3 e-series robot with the staging branch. 

When starting the controller against a simulated robot, the robot starts to move slightly over time, you can easily see it when looking at the pose/angles in the simulator. Just as described above. It only seems to be an issue with  `pose_based_cartesian_traj_controller` as I have tested the other controllers where none of them seems to have the same issue. When testing it against a physical robot it also moves slightly over time.

This could indicate that there might be an issue inside the calculation of the joint commands, when the tcp pose is converted to joint positions. If there is no trajectory to follow, it is not necessary to actually calculate any commands, but the command is still calculated. So maybe a fix could be to not calculate a command when there is no trajectory to follow. 

Another reason could be that the actual joint positions, velocities, actual tcp pose and actual tcp speed are used as current position/velocities inside the driver instead of target position/velocities. If the robot is configured with a slightly wrong payload or some kind of external force the actual position can change, but the target wont. This could also resolve in the robot moving slightly over time, when it is supposed to stand still, because the current positions in the driver changes due to the actual position. 

I think both things is worth looking into, but I think the first problem is most likely to solve this issue, because it also is an issue in simulation, where there is no difference between actual and target.

I will try to create a test that can catch both problems today.

fmauch, 2021-07-08:
> If there is no trajectory to follow, it is not necessary to actually calculate any commands, but the command is still calculated. So maybe a fix could be to not calculate a command when there is no trajectory to follow. The main challenge here would be that the hardware interface is not aware whether there is a trajectory running or not. In fact, it does not even know (or care) that the controller offers a trajectory interface. It simply asks the controller where to go in each control cycle.

That's actually not how ROS controller work by default. I could imagine, that we run into numerical issues here. When no trajectory is active, the controller should command the (Cartesian) pose it reads from the robot. This gets transformed to a joint command in [URScript](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/125f4f7a48b299b07e298cf479e3ba6882c3af3a/resources/external_control.urscript#L213). 

Any numerical inaccuracies could lead to a slow drift over time.

gavanderhoorn, 2021-07-08:
@fmauch: I have seen and worked with `hardware_interface`s which don't write to the hw in `write(..)`, based on some predicate.

As long as that state is explicitly modelled (ie: perhaps controllers need to be reset whenever things do become active again), that works quite ok.

You're correct though that the generic controllers in the various `ros_control` packages and also the default way to implement a `hardware_interface` would be what you're referring to.

urmahp, 2021-07-08:
> I could imagine, that we run into numerical issues here. When no trajectory is active, the controller should command the (Cartesian) pose it reads from the robot. This gets transformed to a joint [...]

I mixed up the controllers when testing it yesterday, I went ahead and tested the wrong the controller `joint_based_cartesian_traj_controller` instead of the `pose_based_cartesian_traj_controller` Sorry for mixing up the controllers.....

However there is also a problem with `joint_based_cartesian_traj_controller` which is also moving slowly over time. I was referring to that when there is no active trajectory, the updateCommand is called with the current state in the [controller](https://github.com/UniversalRobots/Universal_Robots_ROS_controllers_cartesian/blob/main/cartesian_trajectory_controller/include/cartesian_trajectory_controller/cartesian_trajectory_controller.hpp#L130) The TCP position is transformed into a joint position, which is sent to the robot in the [updateCommand function](https://github.com/UniversalRobots/Universal_Robots_ROS_controllers_cartesian/blob/main/cartesian_trajectory_controller/include/cartesian_trajectory_controller/control_policies.hpp#L278-L305). I think that's the calculation that gives the issue due to numerical issues as you pointed out, when there is no active trajectory.

With the `joint_based_cartesian_traj_controller` it is an issue both in simulation and on the real robot. I haven't tested the `pose_based_cartesian_traj_controller` on a real robot, and I wont have access to one any time soon. But with  `pose_based_cartesian_traj_controller` it is not moving slowly over time in simulation, so that seems to only be a problem on a real robot. Therefore it is either a problem with numeric values as you pointed out or it could also be a problem with actual and target positions.

stefanscherzinger, 2021-07-28:
Thanks @urmahp for testing this and having a deeper look.

Interestingly, I could reproduce the drift with both the `joint_based_cartesian_traj_controller` *AND* the `pose_based_cartesian_traj_controller` in simulation.

I went ahead with @urmahp 's suggestion and made the controller's update() open loop [in this PR](https://github.com/UniversalRobots/Universal_Robots_ROS_controllers_cartesian/pull/5). This fixes the drift in both controllers (in simulation).

I would like to add that the RobotHW will still send the content of the *command* buffer on each cycle, thus not behaving differently than what could be expected from *normal* ROS-control. The difference is that the Cartesian trajectory controller now only updates this buffer if it processes an action goal. Having such an open-loop approach in the controller's idle state seems reasonable in my opinion.

fmauch, 2021-08-03:
I think, we can close this with https://github.com/UniversalRobots/Universal_Robots_ROS_controllers_cartesian/pull/5 being merged.

closed 2021-08-03 by fmauch