UniversalRobots/Universal_Robots_ROS_Driver #394: Is there controler in ur_robot_driver that can replace the topic of /ur_driver/URScript  in ur_modern_drive？

sunnymints, 2021-06-01:
I find the new drive ur_robot_drive offer the controler list:
vel_joint_traj_controller:
scaled_vel_joint_traj_controller:
pos_joint_traj_controller:
scaled_pos_joint_traj_controller:
joint_group_vel_controller:
The  joint_group_vel_controller calls speedj and the function of topic /joint_speed is similar in ur_modern_driver, but who replaces the function of  /ur_driver/URScript  in ur_modern_drive？
And other,the joint_group_position_controller is offered in fmauch_universal_robot file of ur_gazebo which can send the joint_value to robot joint directly. The Universal_Robots_ROS_Driver has offered similar controller for realrobot？

fmauch, 2021-06-01:
URScript commands can be sent to the topic [`script_command`](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#script_command-std_msgsstring). Note that the robot has to be in ["remote_control"](https://robodk.com/doc/en/Robots-Universal-Robots-How-enable-Remote-Control-URe.html) mode in order to make that work on an E-Series robot.

I wouldn't recommend doing this. You would have to make sure that your incoming position commands come in steady and smoothly. The chances the result is stuttering are quite high. But, of course, you can simply define one yourself in the `controllers.yaml` file. As the driver provides a JointPositionCommand interface, any position-based controller can be used with it.

I'll close this issue as all question have been answered (I think). If you disagree, feel free to reopen or comment further.

closed 2021-06-01 by fmauch

sunnymints, 2021-06-15:
> URScript commands can be sent to the topic [...]

thanks