UniversalRobots/Universal_Robots_ROS_Driver #395: Unstable connection to UR5 using External Control and ROS melodic

qwq321qwq, 2021-06-01:
# Summary
i can't connect PC (Ubuntu 18.04) to UR3(CB3 3.12) by running
roslaunch ur_robot_driver ur5_bringup.launch limited:=true robot_ip:=192.168.1.3.

But it seems something wrong. When i use PolyScope  move to another pose, the part of the error disappear and i don't know why.

*Introduction to the issue*

# Versions
 - ROS Driver version:melodic+ubuntu18.04
 - Affected Robot Hardware Version(s):UR5(cb3   3.12)
 - URCaps Software version(s):external control 1.05

# This is the error
```
roslaunch ur_robot_driver ur5_bringup.launch limited:=true robot_ip:=192.168.1.3
... logging to /home/heshui/.ros/log/ef03f24e-c2d5-11eb-a686-28d2449a6f6c/roslaunch-heshui-ThinkPad-Edge-E440-12985.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://heshui-ThinkPad-Edge-E440:41015/

SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 125
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /hardware_control_loop/loop_hz: 125
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 125
 * /joint_state_controller/type: joint_state_contr...
 * /pos_joint_traj_controller/action_monitor_rate: 20
 * /pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/goal_time: 0.6
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_joint_traj_controller/state_publish_rate: 125
 * /pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /pos_joint_traj_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /robot_status_controller/handle_name: industrial_robot_...
 * /robot_status_controller/publish_rate: 10
 * /robot_status_controller/type: industrial_robot_...
 * /rosdistro: melodic
 * /rosversion: 1.14.11
 * /scaled_pos_joint_traj_controller/action_monitor_rate: 20
 * /scaled_pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/goal_time: 0.6
 * /scaled_pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /scaled_pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
[... 519 lines cut ...]
[ INFO] [1622550913.856112700]: Service available.
[ INFO] [1622550913.856164143]: Waiting for controller list service to come up on /controller_manager/list_controllers
[ INFO] [1622550913.857018357]: Service available.
[ INFO] [1622550914.014468269]: Robot's safety mode is now NORMAL
[ INFO] [1622550914.017631831]: Robot mode is now RUNNING
[INFO] [1622550914.064766]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1622550914.070337]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1622550914.074721]: Loading controller: joint_state_controller
[INFO] [1622550914.078053]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1622550914.082160]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1622550914.086368]: Loading controller: pos_joint_traj_controller
[INFO] [1622550914.086844]: Loading controller: scaled_pos_joint_traj_controller
[INFO] [1622550914.127121]: Loading controller: speed_scaling_state_controller
[INFO] [1622550914.159118]: Loading controller: joint_group_vel_controller
[INFO] [1622550914.174968]: Loading controller: force_torque_sensor_controller
[INFO] [1622550914.175057]: Controller Spawner: Loaded controllers: pos_joint_traj_controller, joint_group_vel_controller
[INFO] [1622550914.182883]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[INFO] [1622550914.191170]: Started controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[ INFO] [1622550932.067632701]: Robot requested program
[ INFO] [1622550932.067873079]: Sent program to robot
[ INFO] [1622550932.231290859]: Robot connected to reverse interface. Ready to receive control commands.

```

I don't know how it going wrong so could I get some suggestions here?

fmauch, 2021-06-01:
Could you please try to run the driver launchfile with the `debug:=true` flag? And when it crashes and you get the gdb prompt, type `bt` and then <kbd>enter</kbd>.

Please post the output of that.

qwq321qwq, 2021-06-02:
Thanks! I added` debug:=true`here.  @fmauch  orz
And it changged as follows:
```
roslaunch ur_robot_driver ur5_bringup.launch limited:=true robot_ip:=192.168.1.3 debug:=true 
... logging to /home/heshui/.ros/log/86ec78a6-c371-11eb-af5e-28d2449a6f6c/roslaunch-heshui-ThinkPad-Edge-E440-2611.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://heshui-ThinkPad-Edge-E440:34663/

SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 125
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /hardware_control_loop/loop_hz: 125
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 125
 * /joint_state_controller/type: joint_state_contr...
 * /pos_joint_traj_controller/action_monitor_rate: 20
 * /pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/goal_time: 0.6
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_joint_traj_controller/state_publish_rate: 125
 * /pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /pos_joint_traj_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /robot_status_controller/handle_name: industrial_robot_...
 * /robot_status_controller/publish_rate: 10
 * /robot_status_controller/type: industrial_robot_...
 * /rosdistro: melodic
 * /rosversion: 1.14.11
 * /scaled_pos_joint_traj_controller/action_monitor_rate: 20
 * /scaled_pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/goal_time: 0.6
 * /scaled_pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /scaled_pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /scaled_pos_joint_traj_controller/state_publish_rate: 125
 * /scaled_pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /scaled_pos_joint_traj_controller/type: position_controll...
 * /scaled_vel_joint_traj_controller/action_monitor_rate: 20
 * /scaled_vel_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_vel_joint_traj_controller/constraints/elbow_joint/trajectory: 0.1
[... 258 lines cut ...]
[New Thread 0x7fffb3fff700 (LWP 2802)]
[INFO] [1622617805.157562]: Loading controller: speed_scaling_state_controller
[tcsetpgrp failed in terminal_inferior: 无效的参数]
[INFO] [1622617805.244281]: Controller Spawner: Loaded controllers: pos_joint_traj_controller, joint_group_vel_controller
[New Thread 0x7fffb37fe700 (LWP 2806)]
[INFO] [1622617805.252402]: Loading controller: force_torque_sensor_controller
[tcsetpgrp failed in terminal_inferior: 无效的参数]
[New Thread 0x7fffb2ffd700 (LWP 2808)]
[INFO] [1622617805.276825]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller

Thread 1 "ur_robot_driver" received signal SIGSEGV, Segmentation fault.
0x00007fffd439db79 in joint_trajectory_controller::TrajectoryBuilder<trajectory_interface::QuinticSplineSegment<double> >::createGoalHandlePtr() const ()
   from /opt/ros/melodic/lib//libjoint_trajectory_controller.so
(gdb) bt
#0  0x00007fffd439db79 in joint_trajectory_controller::TrajectoryBuilder<trajectory_interface::QuinticSplineSegment<double> >::createGoalHandlePtr() const ()
   from /opt/ros/melodic/lib//libjoint_trajectory_controller.so
[... cut ...] r<realtime_tools::RealtimeServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > > >) ()
   from /home/heshui/ur5/devel/lib//libur_controllers.so
#3  0x00007fffc864d762 in joint_trajectory_controller::JointTrajectoryController<trajectory_interface::QuinticSplineSegment<double>, ur_controllers::ScaledPositionJointInterface>::starting(ros::Time const&) ()
   from /home/heshui/ur5/devel/lib//libur_controllers.so
---Type <return> to continue, or q <return> to quit---

```

lmmmmo, 2021-06-04:
I got the same problem these days with Ubuntu18.04 and UR5e. And everything was ok a month ago~

fmauch, 2021-06-07:
As problem seem to arise with the scaled trajectory controller: You could maybe try to rebuild your workspace to rule out any ABI incompatibilities with the existing `joint_trajectory_controller`.

qwq321qwq, 2021-06-10:
> I got the same problem these days with Ubuntu18.04 and UR5e. And everything was ok a month ago~

It just a problem with the net because i can't use rosdep update in my area.  But i still can't solve it because of the net

qwq321qwq, 2021-06-10:
> As problem seem to arise with the scaled trajectory controller: You could maybe try to rebuild your workspace to rule out any ABI incompatibilities with the existing `joint_trajectory_controller`.
Thanks for you help! The problem is that in my area i can't update  http://packages.ros.org/ros/ubuntu/dists/bionic/InRelease and use rosdep update. I think i should close this issue because it is not ROS's fault.

lmmmmo, 2021-06-10:
> > I got the same problem these days with Ubuntu18.04 and UR5e. And everything was ok a month ago~
> 
> It just a problem with the net because i can't use rosdep update in my area. But i still can't solve it because of the net

Oh~~ After I try 'rosdep update' and rebuild my workspace, it works. Thanks！

closed 2021-06-15 by qwq321qwq