UniversalRobots/Universal_Robots_ROS_Driver #397: [Question] How to do real-time (online) path modifications

Thieso, 2021-06-08:
# Summary

I would like to use output from a sensor to adapt the trajectory of the robot in real time. Is that possible? 
I am not sure whether here is the right place to ask or on the MoveIt repo. 

# Issue details
I have a distance sensor which measures the distance between the TCP and a surface and I want this distance to be constant. I plan my trajectory such that it is constant in theory but due to the errors of the kinematic chain it might be that the distance is not quite correct. I then want to use the sensor input to adapt my trajectory during execution to have a constant distance. 
In UR script this is possible using path offsets but I guess it should also be possible in ROS right? 

Any hints on how to achieve this behavior are appreciated.

fmauch, 2021-07-02:
I don't think, this is currently possible. Most ROS controllers are not designed to be reactive.

However, I guess that somebody is doing research on dynamic path replanning which could be used to achieve this.

Apart from that you could try using the [`cartesian_controllers`](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers) suite (Note, this is not the same as [`ros_controllers_cartesian`](http://wiki.ros.org/ros_controllers_cartesian).

The `cartesian_controllers` use a dynamic tcp target point that can be attached to any arbitrary frame. In the past, we've used that to perform screwing operations on a running conveyor belt, for example.

stefanscherzinger, 2021-09-01:
@urrsk Might this be a use case for our discussions on `current` vs `target` joint positions? I could imagine a ROS-based nominal trajectory with an UR-side sensor-based offset correction.

@Thieso What would be your preferred way?
- [ ] ROS-only solution
- [ ] Mixed ROS-Polyscope solution
- [ ] Polyscope solution

Thieso, 2021-09-02:
Hello, 

I actually solved this problem using the Cartesian controllers in the beta driver. I did not have time to test it extensively but it definitely works in for 2D trajectories. 
That was a pure ROS solution which was what I was aiming for.

stefanscherzinger, 2021-09-02:
Alright, thanks for your feedback!

closed 2021-09-02 by stefanscherzinger

urrsk, 2021-09-02:
@stefanscherzinger It turns out that the target joint position is returned after the overlay of e.g. path offset have been applied. So if we should make that possible, we need to introduce a way the ROS driver or other external controls can get the feedback before the overlay. Still think it is a good idea.