UniversalRobots/Universal_Robots_ROS_Driver #399: Ur10e crashes when I open a camera

albertoDiStefano, 2021-06-10:
Hi everyone!

I'm facing a very annoying issue using the universal robot driver for ROS. I'm using a UR10e and a camera of IDS, both are connected with ethernet to my computer (with a switch ethernet) the fact is that every time the driver is running on ROS if I open the camera stream (from ROS or from its application) the robot driver crashes reporting the following:

    [ INFO] [1623311494.470574112]: Connection to reverse interface dropped.
    [ERROR] [1623311494.549006993]: Could not get fresh data package from robot
    [ WARN] [1623311494.598595218]: Failed to read from stream, reconnecting in 1 seconds...
    [ERROR] [1623311494.649393195]: Could not get fresh data package from robot
    [ERROR] [1623311494.749704221]: Could not get fresh data package from robot
    [ERROR] [1623311494.849875293]: Could not get fresh data package from robot
    [ERROR] [1623311494.950023238]: Could not get fresh data package from robot

If I power off and on the pc everything works fine for a bit but then it crashes again.

I really don't know how to solve it, does anyone know how to do it? Thank you in advance

fmauch, 2021-07-02:
Camera streams typically send large amounts of data over the network. As you are using the same network for communication with the robot, timings of your tcp packages can be unreliable. The driver expects to get data from the robot every two milliseconds. If this does not happen, you will see the above message.

We recommend using a separate network interface for the control connection between the ROS pc and the robot.

stefanscherzinger, 2021-08-09:
Please feel free to re-open this issue if the proposed solution does not solve your use case.

closed 2021-08-09 by stefanscherzinger