UniversalRobots/Universal_Robots_ROS_Driver #401: Build problem with beta version

mayday-sauce, 2021-06-10:
# Summary
I am unable to build the beta branch correctly after following the instructions in README.  I am getting the error that the package ur_calibration cannot be processed. I have previously used the melodic branch of the Universal_Robots_ROS_Driver, but have deleted the packages before the installation of beta branch.

Any help is much appreciated!

*Introduction to the issue*

# Versions
 - ROS Driver version: melodic
 - Affected Robot Hardware Version(s): UR10e

# Impact
The command catkin_make_isolated terminates with an error. I am unable to use the workspace.

## Expected Behavior
catkin_make_isolated runs as described in instructions

## Actual Behavior
Output error: 
cd /home/klara/catkin_ws/build_isolated/ur_calibration && /home/klara/catkin_ws/devel_isolated/ur_client_library/env.sh make
[  0%] Built target rosgraph_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target roscpp_generate_messages_cpp
[  0%] Built target roscpp_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target roscpp_generate_messages_nodejs
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_py
[ 25%] Linking CXX executable /home/klara/catkin_ws/devel_isolated/ur_calibration/lib/ur_calibration/calibration_correction
CMakeFiles/calibration_correction.dir/src/calibration_correction.cpp.o: In function `urcl::comm::BinParser::debug()':
calibration_correction.cpp:(.text._ZN4urcl4comm9BinParser5debugEv[_ZN4urcl4comm9BinParser5debugEv]+0x5c): undefined reference to `urcl::log(char const*, int, urcl::LogLevel, char const*, ...)'
CMakeFiles/calibration_correction.dir/src/calibration_correction.cpp.o: In function `urcl::primary_interface::PrimaryParser::parse(urcl::comm::BinParser&, std::vector<std::unique_ptr<urcl::primary_interface::PrimaryPackage, std::default_delete<urcl::primary_interface::PrimaryPackage> >, std::allocator<std::unique_ptr<urcl::primary_interface::PrimaryPackage, std::default_delete<urcl::primary_interface::PrimaryPackage> > > >&)':
calibration_correction.cpp:(.text._ZN4urcl17primary_interface13PrimaryParser5parseERNS_4comm9BinParserERSt6vectorISt10unique_ptrINS0_14PrimaryPackageESt14default_deleteIS7_EESaISA_EE[_ZN4urcl17primary_interface13PrimaryParser5parseERNS_4comm9BinParserERSt6vectorISt10unique_ptrINS0_14PrimaryPackageESt14default_deleteIS7_EESaISA_EE]+0xae): undefined reference to `urcl::log(char const*, int, urcl::LogLevel, char const*, ...)'
calibration_correction.cpp:(.text._ZN4urcl17primary_interface13PrimaryParser5parseERNS_4comm9BinParserERSt6vectorISt10unique_ptrINS0_14PrimaryPackageESt14default_deleteIS7_EESaISA_EE[_ZN4urcl17primary_interface13PrimaryParser5parseERNS_4comm9BinParserERSt6vectorISt10unique_ptrINS0_14PrimaryPackageESt14default_deleteIS7_EESaISA_EE]+0x108): undefined reference to `urcl::log(char const*, int, urcl::LogLevel, char const*, ...)'
calibration_correction.cpp:(.text._ZN4urcl17primary_interface13PrimaryParser5parseERNS_4comm9BinParserERSt6vectorISt10unique_ptrINS0_14PrimaryPackageESt14default_deleteIS7_EESaISA_EE[_ZN4urcl17primary_interface13PrimaryParser5parseERNS_4comm9BinParserERSt6vectorISt10unique_ptrINS0_14PrimaryPackageESt14default_deleteIS7_EESaISA_EE]+0x1f1): undefined reference to `urcl::log(char const*, int, urcl::LogLevel, char const*, ...)'
calibration_correction.cpp:(.text._ZN4urcl17primary_interface13PrimaryParser5parseERNS_4comm9BinParserERSt6vectorISt10unique_ptrINS0_14PrimaryPackageESt14default_deleteIS7_EESaISA_EE[_ZN4urcl17primary_interface13PrimaryParser5parseERNS_4comm9BinParserERSt6vectorISt10unique_ptrINS0_14PrimaryPackageESt14default_deleteIS7_EESaISA_EE]+0x25b): undefined reference to `urcl::log(char const*, int, urcl::LogLevel, char const*, ...)'
CMakeFiles/calibration_correction.dir/src/calibration_correction.cpp.o:calibration_correction.cpp:(.text._ZN4urcl17primary_interface13PrimaryParser5parseERNS_4comm9BinParserERSt6vectorISt10unique_ptrINS0_14PrimaryPackageESt14default_deleteIS7_EESaISA_EE[_ZN4urcl17primary_interface13PrimaryParser5parseERNS_4comm9BinParserERSt6vectorISt10unique_ptrINS0_14PrimaryPackageESt14default_deleteIS7_EESaISA_EE]+0x38a): more undefined references to `urcl::log(char const*, int, urcl::LogLevel, char const*, ...)' follow
collect2: error: ld returned 1 exit status
CMakeFiles/calibration_correction.dir/build.make:166: recipe for target '/home/klara/catkin_ws/devel_isolated/ur_calibration/lib/ur_calibration/calibration_correction' failed
make[2]: *** [/home/klara/catkin_ws/devel_isolated/ur_calibration/lib/ur_calibration/calibration_correction] Error 1
CMakeFiles/Makefile2:241: recipe for target 'CMakeFiles/calibration_correction.dir/all' failed
make[1]: *** [CMakeFiles/calibration_correction.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

stefanscherzinger, 2021-07-01:
@mayday-sauce Sorry for letting this unanswered.

If you are still interested in the beta features, you could try the latest staging branch of the driver. The change of dependencies required is briefly described [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/390#issuecomment-872017473).

fmauch, 2021-08-03:
Closing as a solution has been proposed. Please feel free to comment or reopen if you feel that this is not resolved.

closed 2021-08-03 by fmauch