UniversalRobots/Universal_Robots_ROS_Driver #402: Cannot find controller.yaml in branch calibration_devel

miguelriemoliveira, 2021-06-10:
# Cannot find controller.yaml to set scaled_pos_traj_controller/follow_joint_trajectory

My issue is very similar to #55. 

I launch 

`   roslaunch ur_gazebo ur10e_bringup.launch
`
And then:

`   roslaunch ur10e_moveit_config ur10e_moveit_planning_execution.launch sim:=true
`

and I get in the terminal of the ur10e_moveit_planning_execution the following:

> [WARN] [1623365383.002014252, 110.328000000]: Waiting for scaled_pos_joint_traj_controller/follow_joint_trajectory to come up
> [ERROR] [1623365389.012152958, 116.328000000]: Action client not connected: scaled_pos_joint_traj_controller/follow_joint_trajectory
> 

From what I read in #55, that issue was solved for multiple people with the suggestion in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/55#issuecomment-669733953

The solution was to change the content of file controllers.yaml.

**The problem is: i do not have this file!**

I followed the installation guidelines in the readme, where it says at some point we must install the 
fmauch/universal_robot.git **in particular  branch calibration_devel**, i.e.:

`git clone -b calibration_devel https://github.com/fmauch/universal_robot.git src/fmauch_universal_robot`

Now from what I found I think in this particular branch calibration_devel does not have this file controllers.yaml. The closest thing I find is the ros_controllers,yaml, but I don't think this is it.

I looked around and there is a branch **calibration** that does have this file, but then it also has the ur_msgs which collide with the the one that must be installed because of the noetic distro (https://github.com/ros-industrial/ur_msgs/issues/13)

Thanks in advance.

# Versions
 - ROS Noetic:

fmauch, 2021-06-28:
When using gazebo, this driver is not in the loop anymore!

Therefore, the workaround proposed in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/55#issuecomment-669733953 doesn't help, there. Also, with the current version of calibration_devel this should not be necessary anymore.

If you look at the lauch file, notice that it has a `sim` flag: https://github.com/fmauch/universal_robot/blob/calibration_devel/ur10e_moveit_config/launch/ur10e_moveit_planning_execution.launch

If launched with `sim:=true` it will remap the trajectory action namespace to `/arm_controller/follow_joint_trajectory` which is the one specified in the [gazebo configuration](https://github.com/fmauch/universal_robot/blob/calibration_devel/ur_gazebo/config/ur10e_controllers.yaml#L5) which is obviously wrong as I just realize.

However, as this is not related to the driver, but only https://github.com/fmauch/universal_robot/blob/calibration_devel I'll close this issue here. Feel free to open a new issue on https://github.com/fmauch/universal_robot/blob/calibration_devel

closed 2021-06-28 by fmauch

miguelriemoliveira, 2021-06-30:
Hi @fmauch ,

thank you for the reply. I tried again today and was able to make it work. In my configuration the file 

https://github.com/fmauch/universal_robot/blob/calibration_devel/ur10e_moveit_config/launch/ur10e_moveit_planning_execution.launch

does not work as it is (I am setting the sim flag to true). 

Not sure why but in my gazebo the name of the action is not **/arm_controller/follow_joint_trajectory**, but rather **/pos_joint_traj_controller/follow_joint_trajectory**, so I had to change the launch file like this:

```xml
 <!-- Remap follow_joint_trajectory -->
  <!--  <remap if="$(arg sim)" from="/scaled_pos_joint_traj_controller/follow_joint_trajectory" to="/arm_controller/follow_joint_trajectory"/>-->
  <remap if="$(arg sim)" from="/scaled_pos_joint_traj_controller/follow_joint_trajectory" to="/pos_joint_traj_controller/follow_joint_trajectory"/>
```

This works fine. Thanks again for the help.

Miguel

fmauch, 2021-06-30:
The file you linked has been changed earlier today. As I said in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/402#issuecomment-869763851 the `arm_controller` was wrong, which I fixed in https://github.com/fmauch/universal_robot/commit/5de2e5d5e6245d45953df2f0cbbf5901c939d222

miguelriemoliveira, 2021-07-03:
Hi @fmauch,

Sorry, I did not understand that you had already commited the correct remapping.

Thanks once again.