UniversalRobots/Universal_Robots_ROS_Driver #403: unable to see the model in rviz

Yitian-Xian, 2021-06-15:
Hi all, I am using this package with UR5 robot arm in Ubuntu 18.04 + ROS melodic. I have successfully installed the package and implemented the following two commands: 

roslaunch ur_robot_driver ur5_bringup.launch robot_ip:=169.254.203.110
roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch limited:=true

Everything seemed good, but when I was doing the third command: roslaunch ur5_moveit_config moveit_rviz.launch config:=true , a rviz window would pop up and nothing was in there. In the terminal, actually there was no warning or errors, which really confused me. I have already put a controllers.yaml file in the config folder and fixed the action_ns bit, so everything should be right.

Could anyone tell me what was wrong and how I can fix that? Thank you!

fmauch, 2021-06-15:
Try
```
roslaunch ur5_moveit_config moveit_rviz.launch rviz_config:=$(rospack find ur5_moveit_config)/launch/moveit.rviz
```

gyxx323, 2021-06-28:
I had the same problem. I am using this package with UR3 robot arm in Ubuntu 18.04 + ROS melodic. 
When I run this command
`roslaunch ur3_moveit_config moveit_rviz.launch config:=true`
RVIZ has no graphics, and there is an error that is "Global Status: Error"

And when I run this command
`roslaunch ur3_moveit_config moveit_rviz.launch rviz_config:=$(rospack find ur3_moveit_config)/launch/moveit.rviz`
The UR3 model appears in RVIZ, but its end does not move.

What should I do when I meet such a problem? Thank you!

ljden, 2021-07-10:
@gyxx323  you probably need to change the planning group from endeffector to manipulator

gyxx323, 2021-07-10:
Yes, that's this question. Thank you @ljden

pibaopi, 2021-07-14:
Hi, I am a little puzzled about this sentence. I searched the entire installation package and did not see the controllers.yaml file and I don’t know how to modify the action_ns bit. 
I think this may be the key to the failure. Can you tell me how to modify it ? thank you! @Yitian-Xian